micropython/micropython #10372: ESP8266 https server

loup73, 2022-12-30:
I have an ESP8266 1M with micropython 1.19.1. I'm not able to open a secure socket that can be accessed from a browser. I used the example code from the 1.19.1 repository (/examples/network/http_server_ssl), the secure socket creation command being:
```py
client_s = ssl.wrap_socket(client_s, server_side=True, key=key, cert=cert)
```
The code returns exception no 267 SSL_ERROR_INVALID_CERT_HASH_ALG.
I checked network logs, browser sends TLS1.2 CLIENT HELLO with 17 cipher suites, 0x002F is one of them, server answers SERVER HELLO with fatal alert, unsupported certificate, even though 0x002F seems to be supported in the 1.19.1 version of the repo.

``
Client address: ('192.168.0.42', 52544)
Client socket: `<socket state=3 timeout=-1 incoming=0 off=0>
<_SSLSocket 3fff1e50>
Request:
Exception serving request: -267
``

Is it possible at all to have on ESP8266 a proper https server that can be accessed from a modern browser?

sylvinus, 2023-01-12:
Hello! I have been looking at cipher suites and can try to help...

Are you using MBEDTLS or AXTLS ?

If AXTLS, these comments might be of interest:
https://github.com/micropython/micropython/issues/3198#issuecomment-690840497
https://github.com/micropython/micropython/pull/6433#issuecomment-1314677572

If MBEDTLS:

Did you mean 0x2F (MBEDTLS_TLS_RSA_WITH_AES_128_CBC_SHA) ?

Have you enabled MBEDTLS_DEBUG_C and set mbedtls_debug_set_threshold to 4 in modussl_mbedtls.c to check what's happening ?

jonnor, 2024-09-02:
This issue is missing information needed to reproduce. Please provide details. If not we will have to close the issue.

closed 2025-02-04 by projectgus