micropython/micropython #10386: RP2: _thread crashes with socket after one loop

jcernato, 2023-01-01:
Device: Raspberry Pico_W, vscode with pico-w-go on ubuntu
Task: connect to a network, create a thread connecting and closing a socket to some device in the network:

`
import  network, time, _thread
import usocket as socket

wlan = network.WLAN(network.STA_IF)
wlan.active(True)
wlan.connect(yourSSID, yourKey)
time.sleep(3)
if wlan.isconnected():
    print("WIFI Connected")
else: print("WIFI Not Connected")

addr = (Server_IP, port)
sock = socket.socket()

def test():
    while True:
        sock.connect(addr)
        print("connected")
        time.sleep(1)
        sock.close()
        print("closed")
        time.sleep(3)

_thread.start_new_thread(test, ())
`
results in:
`
>>> Running test.py
 
WIFI Connected
 
>>> connected
closed
Unhandled exception in thread started by <function test at 0x200098f0>
Traceback (most recent call last):
  File "<stdin>", line 17, in test
OSError: [Errno 9] EBADF
`

So the socket is established the first time, in the second loop it failes with EBADF. This crash is quite severe since even a soft-reset does not resolve the issue and every attempt to connect a new socket (also without threading) crashes with EBADF until a full hard-reset is performed.

I have no clue what's going wrong, any suggestions on how to debug this issue?
I tried disabling the garbage collector but this doesn't change anything.

peterhinch, 2023-01-02:
You cannot reconnect a socket once it has been closed: it is kaput. You have to create a new one.

jcernato, 2023-01-02:
That's a good point, sadly it doesn't solve the issue.
Running the while loop in main (and instantiating a new socket each time) runs fine, moving it into a new thread still crashes.

My problem actually arised with [umqtt_simple](https://github.com/micropython/micropython-lib/tree/master/micropython/umqtt.simple) which does create a new socket each time it's connect() function is called, yet it still crashes when used in a thread more than once. (ECONNABORTED)