micropython/micropython #10397: Pi Pico W WLAN set hostname in STA

alexphobby, 2023-01-02:
Please help, cannot set hostrname in Pi Pico W
firmware: 
MicroPython v1.19.1-782-g699477d12 on 2022-12-20; Raspberry Pi Pico W with RP2040

code:
wlan = network.WLAN(network.STA_IF)
wlan.config(hostname="mypicow")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: unknown config param

fivdi, 2023-01-02:
See https://github.com/micropython/micropython/issues/8906

th3w, 2023-01-02:
I am also interested in this option for the Pico_W. Reading #8906 I found out the following:
In [ issue #8906](https://github.com/micropython/micropython/issues/8906) the required changes were discussed to set a hostname in the cyw43 driver  code under [/driver/cyw43.](https://github.com/micropython/micropython/tree/master/drivers/cyw43) and in /extmod/network_cyw43.
Also the resulting PR's [#8918](https://github.com/micropython/micropython/pull/8918) and [#9058](https://github.com/micropython/micropython/pull/9058)/[#9086](https://github.com/micropython/micropython/pull/9086) changes the cyw43 code in /driver/cyw43/

Currently, however, the rp2 build for Pico_W seems to use the version in submodul https://github.com/georgerobotics/cyw43-driver/tree/2cf328d9e41603405a037a29e081a7d30dd519e6 at  /lib/cyw43-driver

(See CMakelist.txt https://github.com/micropython/micropython/blob/master/ports/rp2/CMakeLists.txt#L243-L247 ) 

```
if (MICROPY_PY_NETWORK_CYW43)
    string(CONCAT GIT_SUBMODULES "${GIT_SUBMODULES} " lib/cyw43-driver)
    if((NOT (${ECHO_SUBMODULES})) AND NOT EXISTS ${MICROPY_DIR}/lib/cyw43-driver/src/cyw43.h)
        message(FATAL_ERROR " cyw43-driver not initialized.\n Run 'make BOARD=${MICROPY_BOARD} submodules'")
    endif()
```

th3w, 2023-01-05:
I found out that, at least when building your own Pico_W firmware  from the actual Micropython Master branch, you can set a default hostname  with the flag CYW43_HOST_NAME.

I have simply added in  ports/rp2/boards/PICO_W/mpconfigboard.h  the line
`#define CYW43_HOST_NAME "uPY_PICOW"`

jimmo, 2023-02-01:
See https://github.com/micropython/micropython/pull/10635

jonnor, 2024-09-01:
This should now have been implemented in the above merge request. Proposing to close this issue.

dpgeorge, 2024-09-02:
Yes, you can now use `network.hostname("name")` to set it.

closed 2024-09-02 by dpgeorge

magnus-0, 2025-08-22:
This is apparently not possible yet:
- dhcp_hostname is an unknown config parameter
- hostname as config parameter is recognized, but not effectively applied
- network.hostname() same thing, no true effect on the hardware.

End results keeps being **no** hostname is being exposed (also not the default CYW43 hostname)