micropython/micropython #10402: printing to io.StringIO subclass leads to seg fault

Carglglz, 2023-01-03:
I'm aware that MicroPython `io` documentation says 

> Note that for efficiency, MicroPython doesn’t provide abstract base classes corresponding to the hierarchy above, and it’s not possible to implement, or subclass, a stream class in pure Python.

Minimal example to reproduce `test_stream.py`:
```python
import io

class TestStream(io.StringIO):
    def __init_(self, alloc_size):
        super().__init__(alloc_size)

test = TestStream(100)
print("Now the seg fault...")
print("hello", file=test)

```

```sh
$ micropython test_stream.py
Now the seg fault...
Segmentation fault: 11
```

But I would expect an `Exception` if this not possible not a crash/seg fault. 
I was able to reproduce this in `unix(MacOS)` and `stm32(pybV1.1)` ports.

Maybe is related to #1067? 
I had a look at `py/mpprint.c` but I cannot see any hint... Is there any tool to debug this types of crashes?
This is the valgrind output:
```==20210== Memcheck, a memory error detector
==20210== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==20210== Using Valgrind-3.19.0.GIT-lbmacos and LibVEX; rerun with -h for copyright info
==20210== Command: micropython test_stream.py
==20210==
==20210== Warning: set address range perms: large range [0x7fff201b6000, 0x80001ffb6000) (defined)
==20210== Warning: set address range perms: large range [0x7fff201b6000, 0x7fff7fb4a000) (defined)
==20210== Warning: set address range perms: large range [0x7fff8e41a000, 0x7fffc01b6000) (noaccess)
==20210== Warning: set address range perms: large range [0x7fffc01b6000, 0x7fffe307a000) (defined)
==20210== Warning: set address range perms: large range [0x7fffe307a000, 0x7fffffe00000) (noaccess)
Now the seg fault...
==20210== Invalid read of size 1
==20210==    at 0x10002C057: ??? (in /usr/local/bin/micropython)
==20210==    by 0x10002E711: ??? (in /usr/local/bin/micropython)
==20210==    by 0x10002E7DB: ??? (in /usr/local/bin/micropython)
==20210==    by 0x10002BB4F: ??? (in /usr/local/bin/micropython)
==20210==    by 0x100030D6F: ??? (in /usr/local/bin/micropython)
==20210==    by 0x100023DF5: ??? (in /usr/local/bin/micropython)
==20210==    by 0x100034D18: ??? (in /usr/local/bin/micropython)
==20210==    by 0x100023FDE: ??? (in /usr/local/bin/micropython)
==20210==    by 0x100078425: machine_time_pulse_us (in /usr/local/bin/micropython)
==20210==    by 0x100077DD4: machine_time_pulse_us (in /usr/local/bin/micropython)
==20210==    by 0x7FFF204DFF3C: ??? (in /dev/ttys016)
==20210==    by 0x1: ???
==20210==  Address 0x19 is not stack'd, malloc'd or (recently) free'd
==20210==
==20210==
==20210== Process terminating with default action of signal 11 (SIGSEGV)
==20210==  Access not within mapped region at address 0x19
==20210==    at 0x10002C057: ??? (in /usr/local/bin/micropython)
==20210==    by 0x10002E711: ??? (in /usr/local/bin/micropython)
==20210==    by 0x10002E7DB: ??? (in /usr/local/bin/micropython)
==20210==    by 0x10002BB4F: ??? (in /usr/local/bin/micropython)
==20210==    by 0x100030D6F: ??? (in /usr/local/bin/micropython)
==20210==    by 0x100023DF5: ??? (in /usr/local/bin/micropython)
==20210==    by 0x100034D18: ??? (in /usr/local/bin/micropython)
==20210==    by 0x100023FDE: ??? (in /usr/local/bin/micropython)
==20210==    by 0x100078425: machine_time_pulse_us (in /usr/local/bin/micropython)
==20210==    by 0x100077DD4: machine_time_pulse_us (in /usr/local/bin/micropython)
==20210==    by 0x7FFF204DFF3C: ??? (in /dev/ttys016)
==20210==    by 0x1: ???
==20210==  If you believe this happened as a result of a stack
==20210==  overflow in your program's main thread (unlikely but
==20210==  possible), you can try to increase the size of the
==20210==  main thread stack using the --main-stacksize= flag.
==20210==  The main thread stack size used in this run was 8388608.
==20210==
==20210== HEAP SUMMARY:
==20210==     in use at exit: 0 bytes in 0 blocks
==20210==   total heap usage: 0 allocs, 0 frees, 0 bytes allocated
==20210==
==20210== All heap blocks were freed -- no leaks are possible
==20210==
==20210== For lists of detected and suppressed errors, rerun with: -s
==20210== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)
Segmentation fault: 11
```
Not sure if this helps...

stinos, 2023-01-04:
> Maybe is related to #1067?
I had a look at py/mpprint.c but I cannot see any hint... Is there any tool to debug these types of crashes?

Not related to that issue, this one is really because as the paragraph you quoted says "it's not possible" which practically means "undefined behavior because a piece of code assumes a pointer points to an object of type X but you are passing it a pointer to an object of type Y" (or something like that - didnt check in detail).

Since it can be reproduced on the unix port you could run it under the debugger, then work your way up the stacktrace; you'll see that in objstringio.c in 
```
STATIC mp_uint_t stringio_write(mp_obj_t o_in, const void *buf, mp_uint_t size, int *errcode) {
    (void)errcode;
    mp_obj_stringio_t *o = MP_OBJ_TO_PTR(o_in);
    check_stringio_is_open(o);

    if (o->vstr->fixed_buf) {
        stringio_copy_on_write(o);
    }
```

`o` doesn't actually to point to a properly initialized `mp_obj_stringio_t` and hence `o->vstr` is some random memory location and hence the segfault.

Carglglz, 2023-01-04:
@stinos Thanks for the explanation, that makes sense, I didn't know "it's not possible" could include a crash/seg fault. But if this is expected then I guess it is not a bug.
FYI I was trying to implement a ring buffer using `io.StringIO` to avoid writing more than `alloc_size` and use it with `print` and `sys.print_exception` but I found I a workaround, so I opened this bug report just in case it was a real issue, then I guess we can close it now.

closed 2023-01-04 by Carglglz

Carglglz, 2023-01-04:
@dlech I don't know, but I've just added that to `variants/standard/mpconfigvariant.h` 
`#define MICROPY_BUILTIN_METHOD_CHECK_SELF_ARG (1)` 
and the seg fault is still there... 🤷🏼‍♂️

Carglglz, 2023-01-04:
Also any idea how to properly do this?
> Since it can be reproduced on the unix port you could run it under the debugger, then work your way up the stacktrace; you'll see that in objstringio.c in

I'm trying to use gdb, but I'm not sure I'm doing it right, the stack backtrace is not helpful at all...?
```
$ gdb micropython
GNU gdb (Ubuntu 9.2-0ubuntu1~20.04.1) 9.2
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from micropython...
(No debugging symbols found in micropython)
(gdb) run test_stream.py
Starting program: /home/cgg/.local/bin/micropython test_stream.py
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Now the seg fault...

Program received signal SIGSEGV, Segmentation fault.
0x000055555559130e in ?? ()
(gdb) bt
#0  0x000055555559130e in ?? ()
#1  0x0000555555593dc0 in ?? ()
#2  0x00005555555941cd in ?? ()
#3  0x0000555555590e9a in ?? ()
#4  0x0000555555595dfa in ?? ()
#5  0x00005555555896eb in ?? ()
#6  0x0000555555598986 in ?? ()
#7  0x000055555558986a in ?? ()
#8  0x00005555555a2af3 in ?? ()
#9  0x00005555555a3321 in ?? ()
#10 0x00007ffff7c60083 in __libc_start_main (main=0x55555556ffc0, argc=2, argv=0x7fffffffe448,
    init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffe438)
    at ../csu/libc-start.c:308
#11 0x000055555557001e in ?? ()
(gdb)
```
Is this how is supposed to be run?

dlech, 2023-01-04:
You need to make a debug build by adding `DEBUG=1` to the `make` args. I also like to set `COPT=-O0` so nothing gets optimized out. I usually change the `BUILD` dir for this too so that I can have a debug and release build at the same time.

stinos, 2023-01-05:
Could also help to use an IDE (CLion/VSCode/Eclipse/...) and debug in the IDE: can be much quicker to navigate around

Carglglz, 2023-01-05:
@stinos @dlech thanks for your explanations, they were on point 👍🏼 .
I've managed to properly debug it with gdb:

```
(gdb) run test_stream.py
Starting program: /home/cgg/.local/bin/micropython test_stream.py
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Now the seg fault...

Program received signal SIGSEGV, Segmentation fault.
stringio_write (o_in=0x7ffff7a3f540, buf=0x7ffff7a3f094, size=5, errcode=0x7fffffffdc7c)
    at ../../py/objstringio.c:83
83	    if (o->vstr->fixed_buf) {
```
@stinos It was exactly where you said 👀 

Then I introduced these changes for type checking
```diff --git a/py/objstringio.c b/py/objstringio.c
index 1a083449b..18114fb06 100644
--- a/py/objstringio.c
+++ b/py/objstringio.c
@@ -80,6 +80,12 @@ STATIC mp_uint_t stringio_write(mp_obj_t o_in, const void *buf, mp_uint_t size,
     mp_obj_stringio_t *o = MP_OBJ_TO_PTR(o_in);
     check_stringio_is_open(o);

+    if (!mp_obj_is_type(o, &mp_type_stringio)) {
+
+        if (!mp_obj_is_type(o, &mp_type_bytesio)) {
+            mp_raise_TypeError(MP_ERROR_TEXT("expecting a StringIO or BytesIO object"));
+        }
+    }
     if (o->vstr->fixed_buf) {
         stringio_copy_on_write(o);
     }
```
Which will raise a `TypeError` instead of the segfault:
```
$ micropython test_stream.py
Now the seg fault...
Traceback (most recent call last):
  File "test_stream.py", line 11, in <module>
TypeError: expecting a StringIO or BytesIO object
```