micropython/micropython #10472: Unable to include the 'uasyncio' module in the Zephyr port

bogdanm, 2023-01-11:
I am trying to test the `uasyncio` module in the Zephyr port. Since it is not enabled by default, I added the following lines in my `mpconfigport.h`:

```
#define MICROPY_PY_ASYNC_AWAIT      (1)
#define MICROPY_PY_UASYNCIO         (1)
```

I have confirmed that `extmod/moduasyncio.c` is compiled, but trying to import it results in an `ImportError`:

```
>>> import uasyncio
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'uasyncio'
```

I see that `uasyncio` has some Python code too and I'm guessing that that code is not included in the firmware image, but I'm 
 fairly new to MicroPython and I'm not sure how I am supposed to accomplish this (LittleFS is enabled and working in the image too, verified by running https://docs.micropython.org/en/latest/zephyr/quickref.html#flash-area). 

Or maybe the issue is completely different. Either way, Google wasn't too helpful on this one.

Zephyr version: 3.1.0
MicroPython SHA1:699477d12d1aa1cb57ff6cfc3f5dc5e97524dbe6
Platform: nRF5340 (app core)
Command line used for building: `west build -b nrf5340dk_nrf5340_cpuapp`

dpgeorge, 2023-01-12:
You'll need to have the Python code from `extmod/uasyncio/` available on the board.  You can either copy this to the board's filesystem, or freeze it into the firmware like this:
```diff

diff --git a/ports/zephyr/CMakeLists.txt b/ports/zephyr/CMakeLists.txt
index bd5068faf..f5daf162f 100644
--- a/ports/zephyr/CMakeLists.txt
+++ b/ports/zephyr/CMakeLists.txt
@@ -30,6 +30,8 @@ project(micropython)
 set(MICROPY_PORT_DIR    ${CMAKE_CURRENT_SOURCE_DIR})
 set(MICROPY_DIR         ${MICROPY_PORT_DIR}/../..)
 set(MICROPY_TARGET      micropython)
+set(MICROPY_CROSS_FLAGS -march=armv6m)
+set(MICROPY_FROZEN_MANIFEST ${MICROPY_PORT_DIR}/manifest.py)
 
 include(${MICROPY_DIR}/py/py.cmake)
 include(${MICROPY_DIR}/extmod/extmod.cmake)
diff --git a/ports/zephyr/mpconfigport.h b/ports/zephyr/mpconfigport.h
index 71836768a..e615a58a8 100644
--- a/ports/zephyr/mpconfigport.h
+++ b/ports/zephyr/mpconfigport.h
@@ -91,7 +91,7 @@
 #define MICROPY_PY_ZEPHYR           (1)
 #define MICROPY_PY_ZSENSOR          (1)
 #define MICROPY_PY_SYS_MODULES      (0)
-#define MICROPY_LONGINT_IMPL (MICROPY_LONGINT_IMPL_LONGLONG)
+#define MICROPY_LONGINT_IMPL (MICROPY_LONGINT_IMPL_MPZ)
 #define MICROPY_FLOAT_IMPL (MICROPY_FLOAT_IMPL_FLOAT)
 #define MICROPY_PY_BUILTINS_COMPLEX (0)
 #define MICROPY_ENABLE_SCHEDULER    (1)
```
And add a file called `manifest.py` to the `ports/zephyr/` directory containing:
```py
include("$(MPY_DIR)/extmod/uasyncio")
```

bogdanm, 2023-01-12:
Thanks @dpgeorge! That worked eventually. I had to do a few more things:

- The build system complained that `micropython-lib` is not configured and suggested that I should run `make submodules`. However, that doesn't work in the Zephyr port, so I ran it in `ports/stm32` instead. It pulled in a bunch of dependencies that I don't need, but it also fixed this issue.
- I had to enable the uselect module by adding this line in `mpconfigport.h`:

```
#define MICROPY_PY_USELECT          (1)
```

I have a question related to the changes that you proposed above: initially I set `MICROPY_CROSS_FLAGS` to `armv8m` since that is the correct architecture for the nRF5340, but then I switched back to `armv6m` when I realized that the cross compiler doesn't support `armv8m`. Do you think this is a problem in practice? I know that there are differences between `armv6m` and `armv8m`, but I don't know how relevant they are to the cross compiler. I'm guessing not very relevant, but I'd like a confirmation.

Thank you!

dpgeorge, 2023-01-12:
> initially I set `MICROPY_CROSS_FLAGS` to `armv8m` since that is the correct architecture for the nRF5340, but then I switched back to `armv6m` when I realized that the cross compiler doesn't support `armv8m`. Do you think this is a problem in practice?

`armv6m` is a subset of the newer architectures so using that will work on all ARM Thumb CPUs.  For the nRF5340 you could use `armv7m` to get better code generation (and longer jumps).  But none of this really matters much because it's only relevant when you have `@micropython.native` and `@micropython.viper` and `@micropython.asm_thumb` in your .py files.

> No idea if it's right or not

A blank `MICROPY_EVENT_POLL_HOOK` will probably be OK, but you could also try defining it to `k_yield()`, it may be more efficient.

MarcelKr, 2024-01-25:
I'm trying to integrate asyncio in zephyr 3.5 (I used [the changes from this PR](https://github.com/micropython/micropython/pull/9335) to update the port) and am getting the following error during build. I found other issues about the `mp_qstr_frozen_const_pool` but they weren't really helpful. Do you guys have an idea what needs to be done to resolve this? I tried to understand the issue by analyzing other ports but wasn't able to solve it.

I'm using Zephyr 3.5.0 and micropython 1.22.1. (Yeah, I know, zephyr 3.5 is unsupported by micropython however the supported version 3.1 is long EOL)

```
[7/398] Generating /home/marcel/tonies-next-gen/next-gen/modules/micropython/mpy-cross/build/mpy-cross
FAILED: /home/marcel/tonies-next-gen/next-gen/modules/micropython/mpy-cross/build/mpy-cross 
cd /home/marcel/tonies-next-gen/next-gen/modules/tonies-sdk/build && make -C /home/marcel/tonies-next-gen/next-gen/zephyr/../modules/micropython/mpy-cross
make: Verzeichnis „/home/marcel/tonies-next-gen/next-gen/modules/micropython/mpy-cross“ wird betreten
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
LINK build/mpy-cross
/usr/bin/ld: build/py/qstr.o: warning: relocation against `mp_qstr_frozen_const_pool' in read-only section `.text.qstr_init'
/usr/bin/ld: build/py/qstr.o: in function `qstr_init':
qstr.c:(.text.qstr_init+0x3): undefined reference to `mp_qstr_frozen_const_pool'
/usr/bin/ld: warning: creating DT_TEXTREL in a PIE
collect2: Fehler: ld gab 1 als Ende-Status zurück
make: *** [../py/mkrules.mk:233: build/mpy-cross] Fehler 1
make: Verzeichnis „/home/marcel/tonies-next-gen/next-gen/modules/micropython/mpy-cross“ wird verlassen
```

bogdanm, 2024-01-25:
> I'm trying to integrate asyncio in zephyr 3.5 (I used [the changes from this PR](https://github.com/micropython/micropython/pull/9335) to update the port) and am getting the following error during build. I found other issues about the `mp_qstr_frozen_const_pool` but they weren't really helpful. Do you guys have an idea what needs to be done to resolve this? I tried to understand the issue by analyzing other ports but wasn't able to solve it.
> 
> I'm using Zephyr 3.5.0 and micropython 1.22.1. (Yeah, I know, zephyr 3.5 is unsupported by micropython however the supported version 3.1 is long EOL)
> 
> If you prefer I can also open a new thread/discussion about this.
> 
> ```
> [7/398] Generating /home/marcel/tonies-next-gen/next-gen/modules/micropython/mpy-cross/build/mpy-cross
> FAILED: /home/marcel/tonies-next-gen/next-gen/modules/micropython/mpy-cross/build/mpy-cross 
> cd /home/marcel/tonies-next-gen/next-gen/modules/tonies-sdk/build && make -C /home/marcel/tonies-next-gen/next-gen/zephyr/../modules/micropython/mpy-cross
> make: Verzeichnis „/home/marcel/tonies-next-gen/next-gen/modules/micropython/mpy-cross“ wird betreten
> Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
> LINK build/mpy-cross
> /usr/bin/ld: build/py/qstr.o: warning: relocation against `mp_qstr_frozen_const_pool' in read-only section `.text.qstr_init'
> /usr/bin/ld: build/py/qstr.o: in function `qstr_init':
> qstr.c:(.text.qstr_init+0x3): undefined reference to `mp_qstr_frozen_const_pool'
> /usr/bin/ld: warning: creating DT_TEXTREL in a PIE
> collect2: Fehler: ld gab 1 als Ende-Status zurück
> make: *** [../py/mkrules.mk:233: build/mpy-cross] Fehler 1
> make: Verzeichnis „/home/marcel/tonies-next-gen/next-gen/modules/micropython/mpy-cross“ wird verlassen
> ```
I'm not using frozen modules, buf if your firmware includes a file system I'd suggest copying the `uasyncio` directory to that FS. Then adjust `sys.path` accordingly in MicroPython to include the FS path and you should be able to import the module.

MarcelKr, 2024-01-25:
Oh yeah, that's a good approach I completely overlooked. I'll try. Thanks for the quick response.

jonnor, 2024-09-02:
The original problem here seems to be solved, as well as several follow-up questions. So I will close this issue now. Please open new specific issues for new problems :)

closed 2024-09-02 by jonnor