micropython/micropython #10477: Stopping BT stack from scheduled function may hang on stm32

andreamerello, 2023-01-12:
I've experienced hang (loop forever inside BT stack) when trying to call `ble.active(False)` from within a scheduled function. This seems to happen when there are alive connections.

I've looked at the code, and I tracked down this to the following reason:

In `bluetooth_nimble_port_shutdown(void)`, `ble_hs_stop()` is first invoked, it attempts to terminate any live connection i.e. it initiates the termination process, and registers a listener to be called later on when the connection eventually has terminated.

Then  `bluetooth_nimble_port_shutdown(void)` polls over a volatile flag that is supposed to be set by the former listener.. But this flag gets never set, and the loop lasts forever.

This seems to be due to the fact that the BLE stack doesn't run because we are looping inside a scheduled function, so that the listener never gets invoked.

There is a timeout mechanism for this, which rely on a "npl_callout", but it seems not working in stm32 port, because those "callouts" seem scheduled by `nimble_port_run(void)` which seems never called on stm32 port.  It even looks like that micropython build succeeds just because the compile optimize-out it, otherwise it fails... 

BTW FWIW my current workaround is to invoke `mp_bluetooth_hci_poll()` inside the above loop, after `MICROPY_EVENT_POLL_HOOK` - It seems to work, but I'm not sure it's really OK in all cases - and to check for a timeout in the very same loop; I now have something like this

```
    while (mp_bluetooth_nimble_ble_state !=  MP_BLUETOOTH_NIMBLE_BLE_STATE_OFF) {
        MICROPY_EVENT_POLL_HOOK;
        mp_bluetooth_hci_poll();
        if (mp_hal_ticks_ms() - initial_time > 1000)
            ble_hs_stop_done_workaround();
     }
```

`ble_hs_stop_done_workaround()` is just a wrapper around  `ble_hs_stop_done()`

robert-hh, 2023-04-15:
I ran into the same problem when adding BLE to the SAMD port, and @iabdalkader found the same issue at the RP2 port, which is not surprising since it uses the same software stack. Looking into the code when it happens I know what's going on, but do not really understand why it gets there. 

The state is, that it sticks in this function `ble_hs_stop_done()` at the very end in a loop processing handlers to be called when a BT shutdown is complete. That is the loop:
```
    SLIST_FOREACH(listener, &slist, link) {
        listener->fn(status, listener->arg);
    }
```
slist refers to a simple linked list called `ble_hs_stop_listeners`, and the SLIST_FOREACH macro just iterates  through the list. When the lock-up occurs, the list is corrupted, in that
for one element of the list the `next` pointer points to itself. When the function `ble_hs_stop_done()` starts, it resets the list head before it processes the list, such that after the function terminates, the list is empty. Elements are added to the list using the function `ble_hs_stop_register_listener()`, which is called by `mp_bluetooth_nimble_port_shutdown()`, being called by mp_bluetooth_deinit(). The list element inserted by `mp_bluetooth_nimble_port_shutdown()` is a single static node. If that node is added twice to the list, the list `ble_hs_stop_listeners` ends up with a self-referring node. BUT, `mp_bluetooth_nimble_port_shutdown()` is only called once, not twice, at least in a single rush between mp_bluetooth_init() and mp_bluetooth_deinit().  The bug is very timing sensitive. If the code timing changes, it may disappear and reappear later. It happens when I run the BLE test suite
from micropython/tests, mostly because the a lot of BLE activity happens.

I tried various approaches to fix it, but came up with two solutions to avoid that state:

a) Change the function `ble_hs_stop_register_listener()` such that it does not insert a node that is already in the list. The list has with MicroPython one element, such that this simple version works:
```
static void
ble_hs_stop_register_listener(struct ble_hs_stop_listener *listener,
                              ble_hs_stop_fn *fn, void *arg)
{
    BLE_HS_DBG_ASSERT(fn != NULL);

    listener->fn = fn;
    listener->arg = arg;
    if (listener != SLIST_FIRST(&ble_hs_stop_listeners)) {
        SLIST_INSERT_HEAD(&ble_hs_stop_listeners, listener, link);
    }
}
```
A better version would check the whole list. That seems to be an acceptable fix, only that it is in the BT-nimble stack.

b) Do not insert anything into the list, but use a different flag to tell is  `ble_hs_stop_done()` finished. That change replaces ` mp_bluetooth_nimble_port_shutdown()` in bluetooth_nimble.c, making it's own callback obsolete. 
```
void mp_bluetooth_nimble_port_shutdown(void) {
    DEBUG_printf("mp_bluetooth_nimble_port_shutdown (nimble default)\n");
    // By default, just call ble_hs_stop directly and wait for the stack to stop.

    mp_bluetooth_nimble_ble_state = MP_BLUETOOTH_NIMBLE_BLE_STATE_STOPPING;

    ble_hs_stop(NULL, NULL, NULL);

    while (ble_hs_enabled_state != BLE_HS_ENABLED_STATE_OFF) {
        MICROPY_EVENT_POLL_HOOK
    }
    mp_bluetooth_nimble_ble_state = MP_BLUETOOTH_NIMBLE_BLE_STATE_OFF;
}
```
But that method breaks the layering of the code.

I let the test suite run repeatedly for hours with both versions, and it did not lock up, when it otherwise did within minutes.

robert-hh, 2023-04-16:
I added a PR to micropython/mynewt-nimble preventing duplicate entries into the stop_listeners list. https://github.com/micropython/mynewt-nimble/pull/2. I hope someone will care.