micropython/micropython #10485: Enable MBEDTLS_GCM_C by default

sylvinus, 2023-01-12:
Hello,

Today on `master`, the UNIX port (among others I haven't tested) is not able to connect to most TLS endpoints because it lacks support for [GCM](https://en.wikipedia.org/wiki/Galois/Counter_Mode).

Indeed, GCM is required for what seem to be by far the 4 most popular Cipher Suites on the web currently:
https://scotthelme.co.uk/top-1-million-analysis-june-2022/

This can be easily tested by this script: ( [SSLLabs for reference](https://www.ssllabs.com/ssltest/analyze.html?d=en.wikipedia.org&s=198.35.26.96&latest) )
```
import ussl
import socket

ai = socket.getaddrinfo("en.wikipedia.org", 443)
s = socket.socket(ai[-1][0])
s.connect(ai[-1][-1])
s = ussl.wrap_socket(s, server_hostname="en.wikipedia.org")
s.write(b"GET / HTTP/1.0\r\nHost: en.wikipedia.org\r\n\r\n")
print(s.read(1024))
s.close()
```

This will fail with `OSError: (-30592, 'MBEDTLS_ERR_SSL_FATAL_ALERT_MESSAGE')` unless `MBEDTLS_GCM_C` is defined in `extmod/mbedtls/mbedtls_config_common.h`.

For some reason, it seems to be enabled only in the `rp2` port.

I know there will be some consequences on the flash size, but if micropython ships with TLS support it makes sense to support today's web. Actually, there could probably be a few flags for known weak protocols to be disabled if we want to recoup the increase due to enabling GCM!

Thanks!

dpgeorge, 2023-01-18:
OK, we probably should enable `MBEDTLS_GCM_C` on all ports that use mbedTLS.  That will make them all consistent (with rp2) and be able to access more sites.

> Actually, there could probably be a few flags for known weak ciphers to be disabled

This was already done recently, see 68f166dae9ad6dfd94038d5f4394defbb44238af and previous commit.

If you know of any others that can be disabled, please comment.

sylvinus, 2023-01-18:
@dpgeorge great! How to feel confident about disabling more? Should there be an automated test on the top N websites? Probably TLS is also used to connect to other types of endpoints (APIs?) we must continue to support?

dpgeorge, 2023-01-19:
> How to feel confident about disabling more? Should there be an automated test on the top N websites?

A lot of MicroPython code is intended to run reliably for many years (eg in an embedded setting) and so there will probably be quite a lot of legacy/old ciphers that are needed.  So I'm not sure how to approach this... we don't currently have a policy on what sites are guaranteed to work and maybe we should.

In the end, a user can always enable ciphers that they need for their project (at the C level, and rebuild firmware), if they aren't enabled by default.

For now, probably we can just enable GCM in the common `extmod/mbedtls/mbedtls_config_common.h` file and leave it at that.

dpgeorge, 2024-07-02:
Done in 838794ebccb0536606bc8e67fda2be5a47c2dffd

closed 2024-07-02 by dpgeorge