micropython/micropython #10508: Timeout support for websocket/webrepl

jimmo, 2023-01-16:
Reported here: https://github.com/orgs/micropython/discussions/10444

Webrepl constructs a websocket with `BLOCKING_WRITE` set. There's no timeout, so prints can hang.

jonnor, 2024-09-02:
Agree this would be good. Anyone interested in taking this on?

agatti, 2024-09-12:
For what it's worth, an implementation of timeouts on websockets is here: https://github.com/micropython/micropython/commit/51b7a14dec7d022a55b0368d5fdcd4b5dd3d8345 - works with the simple tests I've made, but I haven't got anything more substantial to test this against.

@jonnor The webrepl code is in another repository, https://github.com/micropython/micropython-lib/, so maybe this should be specific for websockets.  If this ever gets submitted as a PR and merged (hard to say if I don't have any decent test cases ready to run), perhaps a separate issue should be filed in the other repository as it's specific for the non-native part of webrepl (as in, not extmod/modwebrepl.c).