micropython/micropython #10512: Asyncio.gather error w/PR Proposal

stephanelsmith, 2023-01-16:
Gathering tasks in a fairly large project, I periodically run into:
```
Traceback (most recent call last):
  File "main.py", line 141, in main
  File "uasyncio/core.py", line 1, in run
  File "uasyncio/core.py", line 1, in run_until_complete
  File "uasyncio/funcs.py", line 1, in done
NameError: local variable referenced before assignment
```

I've tracked the error occurs here (current code snippet below):
`if gather_task.data is not _Remove:`

The issue being **gather_task** is not yet defined.  I'm putting together a PR, but wanted to solicit feedback on proposed changes (code snippet below):
- Add "nonlocal gather_task"
- Move the initialization of state and gather_task above the for loop so those variables are guaranteed existing before done is assigned.

With these changes, I'm no longer getting any errors.  If this is sane, I'll put together a PR.

**Proposed**
```
def gather(*aws, return_exceptions=False):
    if not aws:
        return []

    def done(t, er):
        # Sub-task "t" has finished, with exception "er".
        nonlocal state
        nonlocal gather_task  ############################################ ADDED
        if gather_task.data is not _Remove:
            # The main gather task has already been scheduled, so do nothing.
            # This happens if another sub-task already raised an exception and
            # woke the main gather task (via this done function), or if the main
            # gather task was cancelled externally.
            return
        elif not return_exceptions and not isinstance(er, StopIteration):
            # A sub-task raised an exception, indicate that to the gather task.
            state = er
        else:
            state -= 1
            if state:
                # Still some sub-tasks running.
                return
        # Gather waiting is done, schedule the main gather task.
        core._task_queue.push(gather_task)

    ts = [core._promote_to_task(aw) for aw in aws]
    gather_task = core.cur_task  ############################################ ADDED
    state = len(ts)                       ############################################ ADDED
    for i in range(len(ts)):
        if ts[i].state is not True:
            # Task is not running, gather not currently supported for this case.
            raise RuntimeError("can't gather")
        # Register the callback to call when the task is done.
        ts[i].state = done

    # Set the state for execution of the gather.
    #gather_task = core.cur_task   ########################################## REMOVED
    #state = len(ts)                        ########################################## REMOVED
    cancel_all = False
```

**Original**
```
def gather(*aws, return_exceptions=False):
    if not aws:
        return []

    def done(t, er):
        # Sub-task "t" has finished, with exception "er".
        nonlocal state
        if gather_task.data is not _Remove:
            # The main gather task has already been scheduled, so do nothing.
            # This happens if another sub-task already raised an exception and
            # woke the main gather task (via this done function), or if the main
            # gather task was cancelled externally.
            return
        elif not return_exceptions and not isinstance(er, StopIteration):
            # A sub-task raised an exception, indicate that to the gather task.
            state = er
        else:
            state -= 1
            if state:
                # Still some sub-tasks running.
                return
        # Gather waiting is done, schedule the main gather task.
        core._task_queue.push(gather_task)

    ts = [core._promote_to_task(aw) for aw in aws]
    for i in range(len(ts)):
        if ts[i].state is not True:
            # Task is not running, gather not currently supported for this case.
            raise RuntimeError("can't gather")
        # Register the callback to call when the task is done.
        ts[i].state = done

    # Set the state for execution of the gather.
    gather_task = core.cur_task
    state = len(ts)
    cancel_all = False
```

dpgeorge, 2023-01-17:
Thanks for the report (and attempted fix).  I think we need to fully understand what is going wrong here.

I don't understand how `gather_task` can be uninitialised, because the `gather` function does not yield to the scheduler (and hence give a chance for `done` to be called via `task.state`) until after `gather_task` is assigned to.

Unless your code is hitting the `raise RuntimeError("can't gather")` line and that is returning to the caller and leaving things in a bad state?

Said differently, are you able to get a minimal reproduction of the error?

peterhinch, 2023-01-18:
The "can't gather" runtime error occurs when a non-running task is included in the `gather`. See [this line](https://github.com/micropython/micropython/blob/67097d8c2b3887f7b177564c230d0bf34d23bb43/extmod/uasyncio/funcs.py#L93) and its preceding comment.

stephanelsmith, 2023-01-18:
Thank you very much for taking a look and providing feedback.

I didn't think the "NameError: local variable referenced before assignment" while gathering, should be reachable in any case.  But, there's a path. 

I'm having difficulty reproducing the exact situation, it's clearly a corner case in a big project, though I'm keenly interested in addressing and taking seriously every corner case I can find.  In my project, as I've refactored certain elements, I'm not able to reproduce the NameError as I was before.

Nevertheless, I would recommend a minimalist change, defining the dependecies of done() before it's instantiation to definitely prevent a NameError .  Possibly even defining _gather_task_ = None or similar so as to catch an invalid state and throw an exception "can't gather" which would otherwise leave remaining functionality and code-order unaltered until there's a reporducable test case.

stephanelsmith, 2023-01-18:
I reverted back to the original gather function, and just got it again.  Here's the combined error, that is, "can't gather" followed by "NameError".

I think this issue is worth consideration in that the error booted the application out to the asyncio.run() command.  Would be ideal to be able to catch and address it in situ.

```
Traceback (most recent call last):
  File "sqn/sqn.py", line 410, in sqn_coro
  File "uasyncio/funcs.py", line 1, in gather
RuntimeError: can't gather
Traceback (most recent call last):
  File "main.py", line 141, in main
  File "uasyncio/core.py", line 1, in run
  File "uasyncio/core.py", line 1, in run_until_complete
  File "uasyncio/funcs.py", line 1, in done
NameError: local variable referenced before assignment
MicroPython v1.18-1434-g82540672c-dirty on 2023-01-18; IB_ANC0 with ESP32-S3
Type "help()" for more information.
>>>

```

peterhinch, 2023-01-19:
In compiling a task list for `gather` try checking the `.done()` status of each task:
```python
await asyncio.gather(*[t for t in tasks if not t.done()], return_exceptions)
```

stephanelsmith, 2023-01-24:
Will use work around, preventing gathering done task.

closed 2023-01-24 by stephanelsmith