micropython/micropython #10523: esp32: Wifi connection not working on newer ESP32 WROOM boards

robert-hh, 2023-01-18:
### Discussed in https://github.com/orgs/micropython/discussions/10521

<div type='discussions-op-text'>

<sup>Originally posted by **gkzsolt** January 18, 2023</sup>
This is a duplicate of #10377. I am sorry, but cannot help it to be somehow desperate about Micropython not working on the newer WROOM revision 3 dev boards. In a nutshell, network.WLAN().connect works on ESP32D0WDQ6 (revision 1) boards, but not on ESP32D0WDQ5 (revision 3), where a board reset occurs.
My electronics provider says that:
> "Q5" és a "Q6" refer to the IC packaging. Between revision 1 and 3 the latter is the newer, where the manufacturer issued several fixes, which could lead to the errors encountered. 

They provided a link to the [ESP32 Series SoC Errata](https://www.espressif.com/sites/default/files/documentation/esp32_errata_en.pdf), which did not help much. It might be useful to MicroPython developers though. Is any developer following this?

I tried all versions available at the download page, with firmware compiled with ESP-IDF v3.x or v4.x. Up to v1.11 the code just block without returning; higher versions all were resetting the board (tried also nightly builds: e.g. v1.19.1-812-g67097d8c2 (2023-01-18) .bin)

Unfortunately - or well, fortunately - the C/C++ compiled code from Arduino IDE works well, so this is a MicroPython issue.

I would prefer the board with the older chip to use in my projects, but the distributor does not have those. And when you order a board from a distributor, you cannot know the chip version used.

Is there any hope for this to be considered/tackled in the future? What can I do?

Thank you,</div>

robert-hh, 2023-01-18:
I tested that with esp-idf v4.2.2 and v4.4.2, and both fail when the wifi connection starts. I could not decode the traceback yet. The traceback is below. It was built from the master branch for the GENERIC board. According to the map, the PC is inside the function xPortEnterCriticalTimeout().

Backtrace:
```
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4009847a  PS      : 0x00060833  A0      : 0x80095cfd  A1      : 0x3ffce260  
A2      : 0x0000004c  A3      : 0xffffffff  A4      : 0x3ffd7da8  A5      : 0x00000000  
A6      : 0x00000014  A7      : 0x3ffce480  A8      : 0x0000abab  A9      : 0x0000cdcd  
A10     : 0x00060820  A11     : 0x0000abab  A12     : 0xb33fffff  A13     : 0x00060823  
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x0000001e  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x0000004c  LBEG    : 0x400e2a4c  LEND    : 0x400e2a54  LCOUNT  : 0x00000000  

Backtrace:0x40098477:0x3ffce2600x40095cfa:0x3ffce280 0x4015b34a:0x3ffce2c0 0x4015b4ec:0x3ffce2e0 0x4015ae1d:0x3ffce300 0x400fbaab:0x3ffce320 0x400de34f:0x3ffce380 0x400e52be:0x3ffce3b0 0x400e53ed:0x3ffce3d0 0x40084bf5:0x3ffce3f0 0x400de2e2:0x3ffce490 0x400e52be:0x3ffce500 0x40084b62:0x3ffce520 0x400de2e2:0x3ffce5c0 0x400e52be:0x3ffce620 0x400e5360:0x3ffce640 0x40102cad:0x3ffce6d0 0x40102f18:0x3ffce700 0x400e5468:0x3ffce7e0 0x40084e81:0x3ffce820 0x400de2e2:0x3ffce8c0 0x400e52be:0x3ffce910 0x400e52e6:0x3ffce930 0x400f2f92:0x3ffce950 0x400f32cc:0x3ffce9e0 0x400d6eec:0x3ffcea20 
```

robert-hh, 2023-01-19:
The crash seen herehappenss even for my oldest ESP32 boards. It seems more an issue of the changed esp-idf. May be not related to the topic mentioned by the initial post. But it has been discussed already. Only I cannot find the thread.

dpgeorge, 2023-01-19:
You can get a better backtrace using:
```
$ idf.py -p /dev/ttyUSB0 monitor
```
That gives a REPL, then just make the crash occur and it'll print a nice backtrace.

robert-hh, 2023-01-19:
@dpgeorge That worked, only I had to rename `build-GENERIC` to `build`. Now I have a nice backtrace, but the error of the initial post is different.

closed 2023-01-19 by robert-hh

dpgeorge, 2023-01-19:
> I had to rename `build-GENERIC` to `build`

You can pass `-B build-GENERIC` to `idf.py`.

Actually, we should really add a `monitor` target to `ports/esp32/Makefile` to make this easier.

robert-hh, 2023-01-19:
Adding a short delay after `wlan.active()` seems to avoid the issue. That delay can also be done in wlan.active() by a call of `mphal_delay_ms(1)`. Not sure what in mphal_delay_ms() allows the data structures to align.