micropython/micropython #10540: PWM.duty_u16() may not be set to 14 and smaller

TRadigk, 2023-01-21:
Hi dear micropython developers,

testing PWM on my RP2040 board revealed that the output frequency is reset to 50 Hz / 0 Hz (don't have an oscilloscope, just a DMM with rudimentary frequency measurement) when PWM.duty_u16 is called with any value below 14. Frequency in my example is 25000, if that makes any difference. If there is any dependency to frequency I think it might be good to mention this in documentation, maybe port specific.

example code:
```
pin1 = Pin(1)
pwm1 = PWM(pin1)
pwm1.freq(25000)
pwm1.duty_u16(65536) # output frequency is set to 50 or 0 Hz
pwm1.duty_u16(65535) # output frequency is set to 25000 Hz
pwm1.duty_u16(14) # output frequency remains at 25000 Hz
pwm1.duty_u16(13) # output frequency is set to 50 or 0 Hz
```

This is contrary to documentation:
https://docs.micropython.org/en/v1.19.1/library/machine.PWM.html
> Get or set the current duty cycle of the PWM output, as an unsigned 16-bit value in the range **0 to 65535** inclusive.

I could not find any help if this issue is specific to RP2 or to this version of micropython, I did not try any other version.

firmware file name: rp2-pico-20220618-v1.19.1.uf2
Board: Raspberry pico (RP2040)

> MicroPython v1.19.1 on 2022-06-18; Raspberry Pi Pico with RP2040
> Type "help()" for more information.

robert-hh, 2023-01-21:
The machine.PWM implementation has been changed since V1.19.1. Please repeat your test using the most recent daily firmware. At least tI see no 50Hz for a duty_u16 value below 16, but butst of pulses with a 480µs period.  Maybe caused by the design of the frequency divider.
Note that PWM works by dividing a source frequency by an integer. The granularity of the duty cycle decreases  with increasing frequency. At a PWM frequency of 25000 Hz, a granularity of 16 is not surprising.
About PWM.init(): There is a disclaimer here, that not all ports implement all features. https://docs.micropython.org/en/latest/library/index.html. It is obviously not always documented, which port implements which feature.

TRadigk, 2023-01-22:
Hi robert-hh,

thank you for clarification. As said, my DMM is not comparable to an oscilloscope and all I could see was a somewhat like frozen measurement result, often showing something like 50 Hz.
I would like to test latest nightly build, but I found no disclaimer about the rules for nightlys:
* might it brick my board (killing bootloader)
* might it damage my board
* which feature might be not working as expected (I suspect "every feature")
* where do I find documentation about changes (there are numerous actions done with github, but they are not easy to read, or I at least I can't quickly find a summary)
* in summary: are there any sanitary checks aside from unit testing; how "safe" is it to use the nightly?

Since my application won't show any difference if the duty cycle is set to "almost 0" (by 14) or "absolute 0" (by setting 0) I am more likely to clamp my value by these limits and wait for next official release, which might include a way of those values working.
From the technical standpoint I totally understand, that the frequency divider does have a limit and thus might not be able to realize the resulting pulse length. And I am totally fine with it 😊. So far I like my micropython experience ❤️

robert-hh, 2023-01-22:
The nightly builds go through the same test suite as the release build. Only that the latter may have some more python stcripts frozen into flash by default. The upcoming v1.20 release will most likely identical to the nightly build of that date, beside the mentioned changes in the set of frozen files. None of them is expected to damage your board, and the documentation always shows the latest state.
Using a DMM for testing with like 1m long test leads, I guess that the 50Hz you have seen is just a test artifact. Using a simple 10€ logic analyzer would show more relialble results.
About the duty rate resolution. At a c lock frequency of 125MHz and a PWM frequency of 25000 Hz, the effective duty rate resolution is about 12 bit (125e6 / 25e3 = 5000 ~ 4096). The lower 4 bits are ignored. So 15 will always be the same as 0, That will not change with any firmware, and that property is documented. To get a full 16 bit duty cycle resolution at 25kHz, a clock source with a resolution of 610ps (1 /  25000 * 65536 Hz) would be required. That's beyond the capabilities of the RP2040. The best you can get is ~4 ns.

closed 2023-05-24 by TRadigk