micropython/micropython #10543: Circular Import

kdschlosser, 2023-01-21:
OK say I have main.py and another module called test

There is a single function inside of the test module. Inside that function I import 2 variables from main. The test module gets imported into main and other things run setting the variables and whatever else and the very last thing that gets done if that function inside of the test module gets run.

What is happening is the code inside of main.py is getting run a second time when those 2 variables get imported. Those variables are dedicated way at the top of the main.py file.

In CPython I am able to do this without an issue so I am not sure where the problem stems from. In CPython if a module level variable is declared after a class is that uses that variable and an import gets done of only the class when the code runs that is in the class there will be a name error for that module level variable. That is because as soon as the code for the class loads the import mechanism does it's thing any any other code in that module never gets run.

I know the code is being run 2 times because I have allocated a block of DMA memory the first go around and I check to make sure the memory was allocated and I print out a message letting me know it is. If the memory is not allocated I raise an exception. Well I get the message printed out that it is successful and then after more code loads I get the exception for it not being allocated.

I would imagine this is a bug because the code inside of a module that has already been imported should not run more than a single time if it is imported a second time.

I believe I know where the issue is...

I am running my code on an ESP32-WROVER-B

in main.c in the esp32 port you have this code.

```
pyexec_frozen_module("_boot.py");
    pyexec_file_if_exists("boot.py");
    if (pyexec_mode_kind == PYEXEC_MODE_FRIENDLY_REPL) {
        int ret = pyexec_file_if_exists("main.py");
        if (ret & PYEXEC_FORCED_EXIT) {
            goto soft_reset_exit;
        }
    }
```
Does the `pyexec_file_if_exists` function utilize the same mechanics as `import main` or does it use the mechanics of exec where the code is read and run and not imported. This is the only thing I can imagine that would cause that issue. This is because the interpreter has no knowledge of the code in the main.py file being run so when I import the main.py code it gets ran a second time. Plausible? How to correct the issue is beyond me. IDK is the `__name__ == `__main__'` hold true in Micropython like it does in CPython. An imported module that is being imported for the first time I would imagine it would not have `'__main__'` as it's name.

jimmo, 2023-01-24:
> Does the pyexec_file_if_exists function utilize the same mechanics as import main or does it use the mechanics of exec where the code is read and run and not imported.

@kdschlosser Yes exactly this -- main.py is not "imported" at startup and there is no corresponding entry created in `sys.modules` (aka `mp_loaded_modules_dict`).

MicroPython does set `__name__` as you describe - i.e. `if __name__ == "__main__":` works as you expect from CPython.

Fuithermore, you can do `main = __import__("__main__")` to achieve what you want (i.e. get the already loaded "main module" from inside `test.py`).

kdschlosser, 2023-01-24:
FANTASTIC!!! So long as the `if __name__ == '__main__':` works as it should. I had tried moving all of the imports that pointed to main.py so they were inside of functions to at least let main.py fully load. That fell on its face as well.

Ideally this should be fixed so main.py and also boot.py are added into `mp_loaded_modules_dict`. At the very least update the documentation to reflect that those files are not added to the imported modules and if they get imported by any other module it will cause them to be run again. 

It would have saved me a couple of hours worth of debugging my program to figure out what in the world was going on. I was allocating DMA memory and I kept on getting an error stating there wasn't enough. It was driving me bananas tying to figure out why. It was because I was getting an error for the second time around it was doing it. LOL

`pyexec_file_if_exists` would need to return a module or `None` and if not `None` then the module could be added to `mp_loaded_modules_dict`. I don't believe this would be a terribly hard problem to solve. CPython's behavior is the module gets put into `sys.modules` with the name being set to `'__main__'`. I believe if the module gets imported in CPython by elsewhere after it had been run the code will get run a second time. 

Knowing that I am importing a script that was not imported but instead it was ran from a command line I would be able to do `import __main__` tho. I tried this and it doesn't work in MicroPython

Because it would be considered that boot.py would import main.py the proper way to handle this would be the booy.py module would get the `'__main__'` naming and main.py, well...would be `'main'`. In order to keep this consistent if boot.py doesn't exist then an empty module would need to get dynamically created and the empty module would need to be added keeping the import state consistent with the existence of boot.py and main.py.

I happen to have a boot.py because I change the speed of UART0 on my esp32 so it doesn't take forever to upload my files to it. I could rename main.py to something else and simply import it at the end of boot.py. IDK if there would be any ramifications from doing this at all. I have not looked into the code for MicroPython to see if it would cause any ill effects.

jimmo, 2023-01-24:
> CPython's behavior is the module gets put into sys.modules with the name being set to '__main__'. I believe if the module gets imported in CPython by elsewhere after it had been run the code will get run a second time.

Exactly. If you write `import main` in Python, it will re-import main.py. As far as I can tell, MicroPython's current behavior exactly matches CPython, with the only exception that you cannot write `import __main__` only `__import__("__main__")`.

When you said "Inside that function I import 2 variables from main." were you doing `from main import x, y` or `from __main__ import x, y` ?

> Because it would be considered that boot.py would import main.py the proper way to handle this would be the booy.py module would get the `'__main__'` naming and main.py, well...would be `'main'`.

Yeah I think I get what you're saying here, which is that you see it as "boot.py is __main__ and main.py is a regular import".

Conceptually the way micropython sees this is that boot.py is special setup for how to find main.py, and main.py is the actual script being run. i.e. to map it to CPython it would be a bit like

`python --special-startup-to-find-my-main-script=boot.py main.py`

jimmo, 2023-01-25:
One extra note... of course `main.py` is kind of arbritrary here, just like in CPython I can write `python foo.py` and now `foo.py` is my __main__. i.e. the thing that is __main__ is not necessarily named main.py. In some MicroPython ports (e.g. stm32) you can use `pyb.main('foo.py')` in boot.py to change the "main" py.

I guess where I'm going with this is that if you wanted the behavior you describe where your main file was imported, then you could have main.py do only `import app`, and then put all your "main" code in app.py, and then safely do `import app` anywhere else in the codebase.

jimmo, 2023-01-25:
> with the only exception that you cannot write `import __main__` only `__import__("__main__")`.

Actually this is not correct. You can definitely do this in MicroPython. So I think we actually match CPython exactly here?

(I had confused myself because I had to recently use `__import__("__main__")` explicitly in [aiorepl](https://github.com/micropython/micropython-lib/blob/master/micropython/aiorepl/aiorepl.py#L91) but that is for a completely different reason).

kdschlosser, 2023-01-25:
I think you are following me now. Matching CPython exactly. I am pretty sure it can be done.

When you do `python foo.py` there is no `foo` added to sys.modules. But there is `__main__` in sys.modules and that module is foo.py.

One of the things you have to remember is how terribly slow it is to upload to MicroPython. Even with maxing out the CP210x USB to UART at 921600 bps it is still not all that fast and it takes a while to do. The larger the number of files the longer it is going to take.

In my specific use case I have all of the pin definitions in main.py and I need to use 2 of those definitions in another area of the code where it is far easier to import main.py in that module and collect those pin definitions. Now I could make a separate module to hold all of the pin definitions but then we are adding to how long it is going to take to upload and also additional overhead used for the additional module.

This is where it gets tricky. in CPython you would use exec or the like to execute the code. One of the parameters that you can pass to it is a dictionary. I don't know  if `pyexec_file_if_exists` has this ability.

basically in python it would be as follows.

```python

import sys

namespace = {}
sys.modules['__main__'] = namespace
with open('main.py', 'r') as file:
    exec(file.read(), namespace)
```

It has to be done this way so that as the code inside of main.py gets processed the dictionary in sys.modules gets updated as the code runs. You already know that dictionaries are mutable and any changes made to namespace in exec would end up being available to any modules that are imported in the main.py script and in an imported module `import __main__` or `import('__main__')` is used.

I do not know the internal mechanics of MicroPython with respect this this.

I know that one of the goals is to keep functionality as close to CPython as possible with taking cost of doing into consideration. I don't think that the cost to do this would be high. Maybe as low as 4 bytes of memory for the pointer to main.py's global dictionary.

Could add a macro so at compile time it can either be added or not added.

I would think that adding an additional module would have a higher cost then adding the mechanics to put `__main__` in sys.modules so it can be imported. I have not done any legwork to check on that, you probably already know the answer to that tho.

kdschlosser, 2023-01-25:
OK you are correct. `import __main__` does work. when I was not able to do `import main` and I went pocking about the code and came across the code in the first post I didn't see anything that would indicate that main.py is being placed in sys.modules at all. There has to be some kind of back end string comparison being made to check for 'main.py' being passed as a parameter and if it is then the associated `'__main__'` gets added to sys.modules.

Ding ding question here. If there is no main.py file does '`__main__'` still get added to sys.modules? and if it does is there anything contained in that module?

jimmo, 2023-01-26:
Sorry @kdschlosser I'm a bit lost...

As far as I can tell, the _only_ thing we do differently to CPython is that there is no `__main__` in `sys.modules` (because as you say, there is special handling for "__main__" in the import path). We could definitely add this, but I don't see how it would solve your problem because you're not actually using `sys.modules`. Either way `import __main__` works, and will give you access to the top-level interpreter scope (i.e. the scope of boot.py, main.py and the REPL).

> Ding ding question here. If there is no main.py file does '`__main__'` still get added to sys.modules? and if it does is there anything contained in that module?

Not `sys.modules`, but yes, `import __main__` _always_ works, and it imports the top-level scope.

Simple example -- device with empty filesystem:

```py
MicroPython v1.19.1-774-ged58d6e4c on 2022-12-16; PYBv1.1 with STM32F405RG
Type "help()" for more information.
>>> import __main__ as f
>>> a = 1
>>> f.a
1
```

More complicated example -- device has:
boot.py: `py.main('foo.py')`
main.py: `print('main.py'); a= 1`
foo.py: `print('foo.py'); b = 1`

```
MPY: sync filesystems
MPY: soft reboot
foo.py                         <---------- foo.py is running at boot
MicroPython v1.19.1-774-ged58d6e4c on 2022-12-16; PYBv1.1 with STM32F405RG
Type "help()" for more information.
>>> import __main__ as m  # returns the top level scope (which is where foo.py was executed)
>>> import main as m2     # main.py is unrelated to __main__
main.py
>>> import foo as f      # will re-import
foo.py
>>> m.b
1
>>> m2.a
1
>>> f.b
1
```

kdschlosser, 2023-01-28:
There is `'__main__ '`. That is what I am getting at....

If you have in a 'main.py' file the following items

```python

TEST1 = 1
TEST2 = 2
TEST3 = 3

print('this is main loading')

import some_module
```

and then in a file named 'some_module.py' 

```python

from __main__ import TEST2

print('this is some_module loading... TEST2 = ' + str(TEST2))
```

Everything works as it should. `'this is main loading'` is only printed out a single time.
Where as if I had this in 'some_module.py'

```python

from main import TEST2

print('this is some_module loading... TEST2 = ' + str(TEST2))
```

`'this is main loading'` would get printed 2 times

I am not sure where the mechanics are in place to name main.py as `'__main__'` or where it gets stored but it does exist and it does not cause main.py to get run 2 times.

jimmo, 2023-01-30:
> Where as if I had this in 'some_module.py'
> `from main import TEST2`

This is the exact same behavior you'd get from CPython.

```
$ cd /tmp
$ cat > main.py
TEST1 = 1
TEST2 = 2
TEST3 = 3

print('this is main loading')

import some_module
$ cat > some_module.py
from main import TEST2

print('this is some_module loading... TEST2 = ' + str(TEST2))
$ python main.py
this is main loading
this is main loading
this is some_module loading... TEST2 = 2
```

closed 2025-03-03 by kdschlosser