micropython/micropython #10544: Crash with inappropriate use of '@' operator

jepler, 2023-01-22:
This is a minimized version of a test case that came from the aflplusplus fuzz tester:
```
$ ./micropython-coverage  -c '66**66@6'
micropython-coverage: ../../py/objint_mpz.c:301: mp_obj_int_binary_op: Assertion `op == MP_BINARY_OP_DIVMOD' failed.
Aborted
```
I think the code just needs to be changed to raise an unsupported operand types error, like `6@6` does.

dpgeorge, 2023-01-22:
Thanks for the report.  I can confirm the issue.  A simpler case is:
```py
100000000000000000000000000 @ 0
```

dpgeorge, 2023-01-23:
Also similar problem with float disabled but big-int enabled, doing division eg `10000000000000000000000 / 1`.

dpgeorge, 2023-01-23:
See #10553 for a fix.

dpgeorge, 2023-01-23:
Fixed by d387ae3444751a3ba1032bd16243cb2dc741e333

closed 2023-01-23 by dpgeorge