micropython/micropython #10547: Crash on RP2 with pin IRQ's

peterhinch, 2023-01-22:
The problem has appeared on nightly builds since 20th Dec. This build fails: rp2-pico-20230120-unstable-v1.19.1-831-g4f3780a15.uf2. The following is the smallest repro I've come up with so far:
```python
import uasyncio as asyncio
from machine import Pin

class Encoder:

    def __init__(self):
        pin_x = Pin(20, Pin.IN, Pin.PULL_UP)
        pin_y = Pin(17, Pin.IN, Pin.PULL_UP)
        self._x = 0
        self._y = 0
        self._v = 0
        trig = Pin.IRQ_RISING | Pin.IRQ_FALLING
        pin_x.irq(trigger=trig, handler=self._x_cb, hard=False)
        pin_y.irq(trigger=trig, handler=self._y_cb, hard=False)

    def _x_cb(self, pin_x):
        print('x')
        if (x := pin_x.value()) != self._x:  # Crash seems to occur here
            pass

    def _y_cb(self, pin_y):
        print('y')
        if (y := pin_y.value()) != self._y:  # or here
            pass

async def main():
    e = Encoder()
    while True:
        await asyncio.sleep(1)
        print(e._v)

asyncio.run(main())
```
When either pin changes state the ISR prints then the board crashes. If I reduce the ISR's to just a print statement the code runs. Note hard and soft IRQ's behave similarly.

robert-hh, 2023-01-22:
In the irq's you refer to variables `x` and `y`. But I cannot tell where these are created.

peterhinch, 2023-01-22:
They are created by the walrus operator (`:=`). In the full code the value is used later:
```python
    def _x_cb(self, pin_x):
        if (x := pin_x()) != self._x:
            self._x = x
            self._v += 1 if x ^ self._pin_y() else -1
```

robert-hh, 2023-01-22:
Sorry for my spurious comment. In the small font of the E-Mail, the walrus operator looked like `!=`.
Anyhow. It seems like the reference to the function argument pin_x or pin_y kills it. Referring to class member works. 
Anyhow, something in the commit below is the culprit.
`c214c9e64 rp2/machine_pin: Add support for named pins and alternate functions.`
@iabdalkader could you have a look at it?

robert-hh, 2023-01-22:
OK. Found it. Line 426 in machine_pin.c has to be:
`        irq->base.parent = MP_OBJ_FROM_PTR(machine_pin_cpu_pins[pin]);`

instead of:

`        irq->base.parent = MP_OBJ_FROM_PTR(&machine_pin_cpu_pins[pin]);`

machine_pin_cpu_pins[] is already a table of pointers to pin objects.

iabdalkader, 2023-01-22:
@robert-hh I was just about to take a look but if you think that's the issue, feel free to send a PR. Thanks!

robert-hh, 2023-01-22:
@iabdalkader Yes, I tested it with Peter's script and made already a PR.

dpgeorge, 2023-01-23:
Yes it looks like this was introduced in the rp2 named-pins refactor: c214c9e6481749d90a7c487e24afddef061c0d8d

dpgeorge, 2023-01-23:
Fixed by d4a4cde42e80dc2c25c018af200e641b03999e19

closed 2023-01-23 by dpgeorge