micropython/micropython #10548: Windows dev variant build

emile-cronje, 2023-01-22:
Good day

I followed the instructions on https://github.com/micropython/micropython/tree/master/ports/windows in the section 'Variants' to try and build the dev variant. It build successfully, but when I run mp and do import uasyncio, i still get:

ImportError: no module named 'uasyncio'

The manifest.py in the variants/dev folder specifies:
include("$(MPY_DIR)/extmod/uasyncio")

What am I missing?

stinos, 2023-01-23:
> in the section 'Variants'

Just to be clear: this means you're building the msvc port, right? And `import _uasyncio` does work probably?

In any case: the usage of frozen modules via manifest.py (in essence: take .py files and include them in the output executable) is missing from the msvc port and as a result the .py files in extmod/uasyncio are not included in the .exe. What does work is adding the extmod directory to the Python search path either by appending to `sys.path` or by setting the `MICROPYPATH` environment variable or by copying the uasyncio directory to another path which is already in the search path.

stinos, 2023-01-23:
On second thought: this might actually be a bug because 2a9ea69f introduced the ability to freeze modules, I'll check.

stinos, 2023-01-31:
@emile-cronje if this is something urgent check the PR mentioned above, which should fix the issue

jonnor, 2024-09-01:
Issue should have been fixed by above commit. Please open a new issue with more details if issue persists on latest MicroPython in git.

closed 2024-09-01 by jonnor