micropython/micropython #10554: ESP32-C3 board hangs during neopixel.write in loop

RetiredWizard, 2023-01-23:
I'm using the latest nightly GENERIC-C3-USB build on an Adafruit Qt Py C3

MicroPython v1.19.1-831-g4f3780a15 on 2023-01-20; ESP32C3 module with ESP32C3

When running a tight loop that includes a NeoPixel.write() operation the board appears to hang after somewhere between 50 and 200 writes. Adding a time.sleep(.015) before the write() prevents the hang (.01 is not long enough and still hangs).

This code hangs:
```py
import neopixel
import machine
p = neopixel.NeoPixel(machine.Pin(2),1)
rgb = (0,10,10)
for i in range(500):
    if i%10 == 0:
        print(i)
    p[0] = rgb
    p.write()
```
Leaving out the p.write() doesn't cause the board to hang, but the neopixel obviously doesn't light up. The following code runs without the hang, but also takes a few seconds to complete:

```py
import neopixel
import machine
import time
p = neopixel.NeoPixel(machine.Pin(2),1)
rgb = (0,10,0)
for i in range(500):
    if i%10 == 0:
        print(i)
    p[0] = rgb
    time.sleep(.015)
    p.write()
```

I've tested this on a Feather RP2040 and a ESP32-S3 dev kit and neither of those boards hang in the loop without the time delay so this seems to be specific to the ESP32-C3 board.

RetiredWizard, 2023-01-25:
I got the test program to run without hanging by forcing the use of the bitbang bitstream in ports/esp32/machine_bitstream.

```py
void machine_bitstream_high_low(mp_hal_pin_obj_t pin, uint32_t *timing_ns, const uint8_t *buf, size_t len) {
    //if (esp32_rmt_bitstream_channel_id < 0) {
    machine_bitstream_high_low_bitbang(pin, timing_ns, buf, len);
    //} else {
    //    machine_bitstream_high_low_rmt(pin, timing_ns, buf, len, esp32_rmt_bitstream_channel_id);
    //}
}
```
I'll probably stare at the RMT implementation for a while but I don't expect to be able to find the issue since other ESP boards seem to work fine.

I could always submit a PR that forced the bitbang implementation on C3 boards using the `CONFIG_IDF_TARGET_ESP32C3` parameter that's already being used in the bitbang implementation section.

```py
void machine_bitstream_high_low(mp_hal_pin_obj_t pin, uint32_t *timing_ns, const uint8_t *buf, size_t len) {
    #if !CONFIG_IDF_TARGET_ESP32C3
    if (esp32_rmt_bitstream_channel_id < 0) {
        machine_bitstream_high_low_bitbang(pin, timing_ns, buf, len);
    } else {
        machine_bitstream_high_low_rmt(pin, timing_ns, buf, len, esp32_rmt_bitstream_channel_id);
    }
    #else
    machine_bitstream_high_low_bitbang(pin, timing_ns, buf, len);
    #endif
}
```
I've built and tested this change and it does resolve this issue.

RetiredWizard, 2023-01-25:
After focusing on the bitstream function I went back and searched open issues and #8109 seems like it might be related. 

That issue was talking about neostrips with multiple pixels and sure enough the bitbang method appears to break multiple pixel strips. I hooked up a 5 pixel strip and the first pixel didn't set correctly. The other four seem to work fine. I then flashed the original code using the RMT implementation and the full strip behaved normally.....

rdagger, 2024-06-25:
I'm having the same problem on a Qt-Py ESP32-S2.  
15ms was not enough so I increased the delay to 30ms which stopped the freezing.  Unfortunately, there are some parts of my code that are time critical, and I can't add any delays.
I'm running ESP32_GENERIC_S2-20240602-v1.23.0.

Any progress made on this issue?

ricksorensen, 2024-06-27:
See @rdagger discussion #15346.  I switched to the ESP 5.21 compiler/SDK and did not see the problem.  The 5.1 version also seems to work.  Using the ESP 5.04 version I see the hang very quickly.  I wonder how that relates to te @RetiredWizard patch using RMT.

I was using a XIAO ESP32C3 for experimenting, with the latest micropython developments pulled.

ricksorensen, 2024-09-05:
I just ran a simple test with the current releases for both the TinyS3 ('UM_TINYS3-20240904-v1.24.0-preview.286.g042693496.bin') and Generic C3 ('ESP32_GENERIC_C3-20240904-v1.24.0-preview.286.g042693496.bin') on a XIAO ESP32C3.  For each board use the following test code:
```python
import machine
import esp32
import neopixel
pix=neopixel.NeoPixel(machine.Pin(2), 2)
pix[0]=(127,0,128)    # RGB Purple
pix[1]=(0,63,0)           # RGB Green
pix.write()
# check results
# switch to bit-banging
esp32.RMS.bitstream_channel(None)
pix.write()
```
I used a [https://www.amazon.com/gp/product/B09PBGZMNS/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&th=1](https://www.amazon.com/gp/product/B09PBGZMNS/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&th=1) LED strip- driven with 3.3V from MCU.  Data line was GPIO2 for both the TinyS3 and XIAO ESP32C3.

Results were same for both boards:
RMT write:   4(?) LEDs were lit - Purple Green Purple Green
                      note: `pix[2]`, `pix[3]`  give `index out of range`
                                `pix.buf[6]` also gives `index out of range`
Bitbang write: 2 LEDs were lit- White Green
                      note: `pix[0]` --> `(127,0,128)`
                                `pix.buf` --> `bytearray(b'\x00\x7f\x80?\x00\x00')`

I used a logic analyzer to capture the writes.  The neopixel write with RMT does indeed sendout 4 pixels worth of data, while the bitbang write has a peculiar first bit interpreted as a 1 which causes the first pixel to be white.

S3 write of 2 pixels with RMT:
[image]
The timings look correct, but for double the number of pixels.

S3 write of 2 pixels with bit-banging:
[image]
The first bit of first pixel has a long pulse for what should be a zero:
[image]

Here are S3 timings:
```
rmt3: purple,green,purple, green   4 pixels lit
   119.27 uS     # total time
   bit: 800kHz, 1.25uS               High time for 1: 800ns, 0: 400ns

bang: white, green   2 pixels, first has bit 7 set
   60.144 uS     # total time
   bit: 800kHz, 1.25uS              High time for 1: 820ns,  0: 425ns  (single measurement point)

   first bit: 2.06 us total time.        high: 1.11us  low: 0.95 us
   2nd bit: 1.235 us total time.       high: 0.41us   low: 0.825 us
   first led shows as white (0x7f8080) instead of 0x7f0080
```

Here are C3 timings:
```
rmt1: purple,green,purple, green   4 pixels lit
   119.15 uS
   bit: 800kHz, 1.25uS           High time for 1: 800ns, 0: 400ns

bang: white, green   2 pixels, first has bit 7 set
   61.425 uS
   bit: 784kHz, 1.275uS             High time for 1:  790ns  0: 385ns  (single measurement point)

   first bit: 391 uS  total time.      high: 1.66us  low: 1.85 us
   2nd bit:   1.25 uS total time.     high: 0.4us   low: 0.85 us
   first led shows as white (0x7f8080) instead of 0x7f0080

```
I have the full captures of these timings if desired.

I can't imagine that if this was as core (micropython or esp-idf) issue it hasn't been noticed, but i do not see any obvious error in my coding.  I also went back to micropython 1.19 and saw similar results.

Possibly related: #15682,

ricksorensen, 2026-03-16:
I believe the switch to the new RMT implementation for IDF 5 addressed this.

mattytrentini, 2026-03-16:
Based on @ricksorensen's comment it seems like this is resolved so I'll close this for now. If anyone is able to reproduce the problem with a recent build then we can reopen it.