micropython/micropython #10562: Implementing libc's `<string.h>` locally in `string0.c` and `-ftree-loop-distribute-patterns`

josuah, 2023-01-23:
I ran into this GCC bug on a custom nRF port on github builds with `ubuntu-latest` (but not on a local Debian 5.10.140 VM with GCC `10.2.1 20210110`):
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=56888

The suggested fixes are to enable `-fno-tree-loop-distribute-patterns` flag until this gets sorted out:
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=96628#c4
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=56888#c22

The observed behavior is a hard fault, likely due to `memset()`'s body highlighted below being turned into a call to `memset()`:
https://github.com/micropython/micropython/blob/5cf71b55960651fc506df0ac41490f12dd1d63fd/shared/libc/string0.c#L90-L108

The same seems to happen with the startup files, where function call (i.e. to `memset()`) are still not ready.

I will try to offer a reproducer, and eventually a merge request soon.
Feel free to ping me if that does not come fast enough.

dpgeorge, 2023-01-24:
We do have `CFLAGS_BUILTIN` for this kind of thing (so any flags needed for a workaround could be added to that).  Would be good to understand what gcc versions this happens with.

josuah, 2023-01-24:
From a screenshot, this GCC was the one likely coming with brew for a custom build:

```
gcc version 12.2.1 20221205 (Arm GNU Toolchain 12.2.Rel1 (Build arm-12.24))
```

A few months ago, I could reproduce it with github build actions, but do not have the version used at the time.

In my build vm, I do not reproduce the problem:

```
gcc version 10.2.1 20210110 (Debian 10.2.1-6)
```

Thank you for the bits on `CFLAGS_BUILTIN`, this will be more convenient than enabling them globally.

josuah, 2023-01-26:
As seen in [this comment](https://github.com/micropython/micropython/commit/e22b94350889af32527e4ebf3a8323e4eba9280e#r98149552), I try to find the reason for having a local libc in `string0.c` in the first place, as opposed to use i.e. newlib.

Both using newlib and adjusting `CFLAGS_BUILTIN` would fix that issue.

jonnor, 2024-09-02:
Is this still a problem with recent GCC and MicroPython versions? Can anyone recommend/specify a GCC version that *does* work?

If recent compiler versions work, then I propose that we close this issue, as we unfortunately do not have manpower to support every single GCC version.

closed 2025-02-04 by projectgus

josuah, 2025-02-04:
Sorry for the long silence, as I faced more work than I could handle and dropped most communication.
This seems like not being a problem in practice, just mis-configuration on our end.

jonnor, 2025-02-04:
Thanks for the feedback @josuah and hope things are going better now! :)

josuah, 2025-02-04:
Yes that was momentary! Still not quite enough time to upgrade the Zephyr MicroPython port for now 😅