micropython/micropython #10572: Socket is stuck untile reboot

AndreaPici, 2023-01-24:
Hi there,

I have a problem with sockets. I'm using this piece of code to connect and disconnect to a server using socket:

```
class EXAMPLE:

    def __init__(self, slave_ip, slave_port, timeout):
        self.slave_ip = slave_ip
        self.slave_port = slave_port
        self.timeout = timeout
        self._sock = None

    def connect(self):
        created = False
        try:  
            self._sock = usocket.socket(usocket.AF_INET, usocket.SOCK_STREAM, usocket.IPPROTO_TCP)
            self._sock.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)
            self._sock.setblocking(True)
            self._sock.settimeout(self.timeout)
            self._sock.connect(usocket.getaddrinfo(slave_ip, slave_port)[0][-1])
            created = True
            print('Socket created')                
            return created

        except:
            print('Cannot connect socket')
            self.close()
            return created 

    def close(self):
        try:
            if self._sock:
                self._sock.close()
                self._sock = None
            return True
        except:
            raise Exception('Cannot close socket')
```

The problem is that is sometimes stuck and is not able to connect to the server. I see that "settimeout" is done correctly but then stucks during connect. Timeout doesn't affect anything, it will stay in that stuck state for long time.
Actually I'm using an external timer that checks for stuck socket, it starts when socket.connect() is called and is cancelled when socket is created. If not created then a machine.reset() is done. 

After reset everything works correctly.

Please can you help me finding out what's wrong? Thank you very much.

peterhinch, 2023-01-24:
Is it possible that `usocket.getaddrinfo` is blocking? You could test this with print statements:
```python
            print('Getting address')
            a = usocket.getaddrinfo(slave_ip, slave_port)[0][-1]
            print('Got address')
            self._sock.connect(a)
```

dpgeorge, 2023-01-24:
What port and hardware are you running this code on?

dpgeorge, 2023-01-26:
> I'm running the code on a Pycom WIPY 3 board, Pycom Micropython release 1.20.2.r6

Then you have the wrong repository.  This repository is the main MicroPython software, but Pycom have their own variant of MicroPython that supports their boards.  You will have to ask them (more likely their community) for support.

closed 2023-01-26 by dpgeorge