micropython/micropython #10578: ports/rtc: Inconsistencies between ports and the documentation.

robert-hh, 2023-01-24:
It looks like the machine.rtc() API is different for all ports and to the documentation. 

- to start with the good part: In all ports rtc.datetime() matches the documentation.
- rtc.init() differs from the documentation in the item order of the datetime tuple for the ports esp32, samd, mimxrt.
- no rtc.init() at the rp2 port.
- rtc.init() at the stm32 and renesas.ra port just reinit's the RTC device and ignores the function argument.
- rtc.init() for the cc3200 works as documented
- rtc.now() is only available in the cc3200 and mimxrt port, with different datetime tuples.

So it seems as if the documentation of rtc.init() and rtc.now() only matches the cc3200 port. I can make that all consistent, but we should agree first on which class methods should be available. So a simple approach would be:
- keep rtc.datetime() as it is now, resp. add it to the cc3200 port.
- drop rtc.now().
- drop the set date/time functionality of rtc.init() and/or change at least the documentation.

peterhinch, 2023-01-25:
You've probably seen https://github.com/micropython/micropython/pull/5553. Fixing this is long overdue but it will break code.

Should the `rtc.datetime` tuple match the format of `time.localtime`? (This would break even more code, though.)

robert-hh, 2023-01-25:
The section of the documentation which you changed is the only one which matches rtc.datetime(). But all the quickref paragraphs are wrong.My preference is in the post before in the last four bullets. I do not suggest to change rtc.datetime(), since it seems the only part which is consistent over the ports, if implemented at all.

zzzbatmand, 2023-01-25:
How about adding an `rtc.localtime` that follow the `time.localtime` format?
This way, there is a function that follows the `time` format, and don't break any existing code.

robert-hh, 2023-01-25:
Why? rtc.datetime() and time.localtime() use the same hardware clock, if it exists. So there is no benefit in having a rtc,localtime().

zzzbatmand, 2023-01-25:
What I really want, is a simple way of setting the RTC using something like Unix Timestamp.
It is possible to use `time.gmtime` to convert seconds to a datetime format, but this don't fit with the `rtc.datetime` as stated.

An alternative solution is make a function like `rtc.set_epoc` or similar.
Then even libraries that can get the time, can be converted to epoc and used to set the RTC.

robert-hh, 2023-01-25:
See https://github.com/micropython/micropython-lib/tree/master/micropython/net/ntptime

robert-hh, 2023-01-25:
For the SAMD port i had made an extension for boards without RTC, that you could supply a time value as seconds since epoch as argument to time.localtime(). But that's non-standard.

robert-hh, 2023-01-28:
@dpgeorge Do you have a preference about how to proceed with reducing the inconsistencies of machine.RTC. I can at least make an initial PR to fix the documentation of rtc.datetime(), assuming that this method will not be changed.

villeneuve, 2023-05-03:
Hi!
Correct me if I'm wrong, but I see a mistake in the doc for the rp2 RTC. We can read:
rtc.datetime((2017, 8, 23, 2, 12, 48, 0, 0)) # set a specific date and time, eg. 2017/8/23 1:12:48

The code works ok, no problem. But the comment should be read: #... eg. 2017/8/23 12:48

As per the class RTC doc, The 8-tuple has the following format: (year, month, day, weekday, hours, minutes, seconds, subseconds)

So the time set is 12:48:00   not 1:12:48

closed 2024-12-28 by robert-hh