micropython/micropython #10585: ESP32S3 ADC Pins not working

TGiles1998, 2023-01-25:
Hi,

When trying to read an adc value from the adc pins on the esp32s3 I get the following output:

>>> from machine import ADC
>>> adc = ADC(machine.Pin(7, machine.Pin.IN))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: invalid bits

Looking at the C code I notice that there isnt a #if CONFIG_IDF_TARGET_ESP32S3

https://github.com/micropython/micropython/blob/master/ports/esp32/machine_adc.c

Thanks,

Thomas

robert-hh, 2023-01-25:
There is a CONFIG_IDF_TARGET_ESP32S3 in line 70:
https://github.com/micropython/micropython/blob/67fac4ebc53db6337008ba06df7932faec80f57c/ports/esp32/machine_adc.c#L70

TGiles1998, 2023-01-25:
@robert-hh 

Thank you for your reply. Yes thank you I found it too. have you got ADC to work on ESP32S3?

robert-hh, 2023-01-25:
I never tried it.

robert-hh, 2023-01-26:
I just tried it. I can run your code snippet w/o problems, and it returns values vaguely correlated to the input voltage.
`MicroPython v1.19.1-669-gd4b9df176e on 2022-11-05; LOLIN S3 with ESP32S`

mattytrentini, 2023-10-23:
I believe this has been resolved, closing.

closed 2023-10-23 by mattytrentini