micropython/micropython #10590: ffi callbacks are not collected

abiliojr, 2023-01-26:
While working with the UNIX port, I realized that my libcurl wrapper was causing a "memory leak". Upon further investigation, I managed to reduce the code to reproduce the issue to:

```python
import ffi
import gc

def cb():
    pass

for i in range(1000000):
    ffi.callback("v", cb, "") # this is never collected
    if (i % 1000) == 0:
        gc.collect()
        print(gc.mem_alloc())
```

You will see the counter increasing until it runs out of memory.

By negating this condition:
`#if MICROPY_FORCE_PLAT_ALLOC_EXEC`

In file `ports/unix/alloc.c` The problem goes away. The function `ffi_closure_free` has an interesting 'TODO' comment that's been sitting there for 7 years.

The comment in this conditional guarded section reads `save a lot of code size`. How much is a lot?

Micropython version: 1.19
Board: x64 and ARM64 (Linux)

stinos, 2023-01-26:
It could be that it is actually the same issue as in #7273 even though there the opposite problem is reported

abiliojr, 2023-01-26:
Hey @stinos, we were the original reporters of #7273 (in the forum). This one look different. As a side note, we never saw the 7273 issue in Intel or ARM, and this might be the reason why.

stinos, 2023-01-26:
Ah ok :) In that case: you're not using the fix mentioned in the other issue here, but an unmodified MicroPython? And when you say 'the problem goes away' does that actually mean 'the callback is properly garbage collected' or only 'there is no more memory leak' (which could still mean it gets garbage collected when it shouldn't)?

abiliojr, 2023-01-26:
I'm running stock 1.19 . When I say the problem goes away, I mean the garbage gets collected (in my example, it prints a constant number). I ran it over valgrind, and valgrind didn't report any memory leak.