micropython/micropython #10621: RP2: wrong NameError in certain combination of threading, global variables and and single import

rkompass, 2023-01-29:
A global variable reportedly is not there (`NameError: name 'a' isn't defined`) although it should be there.
This occurs in a setting of a recursion test function that hits the limit of 19 recursions on the second thread/core on a RPi2040.
The global variable, `a`, is updated in this function.
Strangely this only occurs if also two other global variables are defined, `sleep_ms` is imported as a single import and the maximum recursion depth exception is catched with try/except.
The code is as follows:
```
import _thread
from time import sleep_ms #, sleep_us  # if more than one functions imported there is no NameError

a = 0
b = 0

def recursionTest():
    global a
    a += 1
    try:
        recursionTest()
    except Exception as errorMsg:
        print(errorMsg)

_thread.start_new_thread(recursionTest,())

c = 0
sleep_ms(10)
print('recursion depth:', a)
```
The output is as follows:
```
Unhandled exception in thread started by <function recursionTest at 0x20008d50>
Traceback (most recent call last):
  File "<stdin>", line 9, in recursionTest
NameError: name 'a' isn't defined
recursion depth: 0
````
If `c = 0`, for example is commented out the output changes to (correct);
```
maximum recursion depth exceeded
recursion depth: 19
```
This was on `MicroPython v1.19.1 on 2022-07-27; Raspberry Pi Pico with RP2040

GuillaumeLeclerc, 2023-04-09:
I have a similar problem and I'm not using recursion. Same uC, also using threading.

Any updates ?

GuillaumeLeclerc, 2023-04-09:
For me it seems that declaring more global variables "fixes" the issue....

rkompass, 2023-04-09:
Could you give code examples? They might help locate the bug.

pdg137, 2023-10-13:
I still see this error on the released `MicroPython v1.21.0 on 2023-10-06; Raspberry Pi Pico with RP2040`.

Almost any change to the code removes the error, for example removing `b`:
```python
import _thread
from time import sleep_ms #, sleep_us  # if more than one functions imported there is no NameError

a = 0

def recursionTest():
    global a
    a += 1
    try:
        recursionTest()
    except Exception as errorMsg:
        print(errorMsg)

_thread.start_new_thread(recursionTest,())

c = 0
sleep_ms(10)
print('recursion depth:', a)
```

Here is the simplest failing version I've been able to create:

```python
import _thread

a = b = None

def test():
    try:
        global a
        a
    except Exception as e:
        print(e)

_thread.start_new_thread(test,())

c = d = None
```

pdg137, 2023-10-13:
Not sure if it is related, but this version gives unpredictable results:

```python
import _thread
from time import sleep_ms

x = y = 1
def test():
    try:
        global a
        a = 1
    except Exception as e:
        print(e)

_thread.start_new_thread(test,())

c = d = None
sleep_ms(10)
print("a is "+str(a))
```

When run from Thonny with F5 sometimes I get this crazy result:

```
>>> %Run -c $EDITOR_CONTENT

MPY: soft reboot
a is __main__
>>> 
```

Apparently `a` might sometimes be set to `__main__` instead of 1??

pdg137, 2023-10-13:
I think the title of this issue should be changed since the recursion part is irrelevant (it never has even a single recursion in the original example).

pdg137, 2023-10-13:
From what I've tried, it looks like a workaround might be to call `sleep_ms(1)` immediately after `start_new_thread()`.  I suspect that whatever it's doing at the beginning with variables is not thread-safe.

rkompass, 2023-10-14:
Nice finding. It's always better to have a more reduced example code eliciting a bug.
Obviously the recursion is unrelated.

> I suspect that whatever it's doing at the beginning with variables is not thread-safe.

Yes, apparently especially if the new thread is in try/except mode.
That introduction of a new global variable directly after starting the thread messes up the global variables.
@pdg137 if you don't mind I will start a new issue and reference this one here. You may do that if you like.
It feels better than renaming this one. Once the remedy is there both issues may be closed, of course.

```py
import _thread
from time import sleep_ms

a = b = 1          # these have to be 2 variables
def test():
    global a
    try:           # the try / except block has to be there
        a = 2
    except Exception as e:
        print(e)

_thread.start_new_thread(test,())
c = 5              # and also this variable has to be here, before the sleep_ms()

sleep_ms(10)
print("a is "+str(a))
```

pdg137, 2023-10-15:
@rkompass of course I don't mind if you start a new issue.