micropython/micropython #10622: SDCard `readblocks()` method crashes pyboard when trying read block -1

victorallume, 2023-01-30:
On a Pyboard D series I've discovered the following bug:

```
>>> import pyb
>>> bd=pyb.SDCard()
>>> bd.power(True)
True
>>> bsize=bd.ioctl(5,None)
>>> nblocks=bd.ioctl(4, None)
>>> buf=bytearray(bsize)
>>> bd.readblocks(0, buf)
True
>>> bd.readblocks(1, buf)
True
>>> bd.readblocks(nblocks-1, buf)
True
>>> bd.readblocks(nblocks, buf)
False
>>> bd.readblocks(-1, buf)
[Pyboard fails to respond until watchdog or manual reset]
```

This is on Micropython 1.18.

`bd.writeblocks(-1, buf)` works fine (i.e. returns `False` since it failed the operation)

jonnor, 2024-09-02:
Hi and thank you for the report. Does this issue still appear in the latest MicroPython, version 1.23 or later?

victorallume, 2024-09-03:
Still appears in v1.23

projectgus, 2025-02-05:
Thanks @victorallume for being patient while someone got around to investigating. I'm able to reproduce on PyBoard V1.1, fix incoming.

dpgeorge, 2025-02-11:
I cannot reproduce this issue.  I tried running the above code on both PYBV11 and PYBD_SF6, with an SD card inserted.  The SD card is 4GB with bsize=512 nblocks=7716864.  I used current master at 11c9656fad716d42a33e9f5bc65596300d141daa .  On both boards the `bd.readblocks(-1, buf)` returns `False`.

projectgus, 2025-02-11:
How odd. I get this every single time on my PYBV11:

```
❯ mpremote a0
Connected to MicroPython at /dev/ttyACM0
Use Ctrl-] or Ctrl-x to exit this shell
MicroPython v1.25.0-preview.283.g11c9656fad on 2025-02-12; PYBv1.1 with STM32F405RG
Type "help()" for more information.
>>> 
>>> 
>>> import pyb
>>> bd = pyb.SDCard()
>>> bd.power(True)
True
>>> buf=bytearray(512)
>>> bd.readblocks(0, buf)
True
>>> bd.readblocks(-1, buf)
(hangs)
```

I've tried skipping the `bd.power(True)` step and/or skipping the successful `bd.readblocks(0, buf)` step and it hangs the same in both cases.

Card specifics:

```
>>> bd.ioctl(5, None)
512
>>> bd.ioctl(4, None)
15122432
>>> bd.info()
(7742685184, 512, 1)
```

Maybe it's specific to particular SD card controller chips?

projectgus, 2025-02-11:
Here's another bit of behaviour I can't really explain:

```
>>> import pyb
>>> bd = pyb.SDCard()
>>> bd.power(True)
True
>>> buf = bytearray(512)
>>> bd.readblocks(0, buf)
True
>>> bd.readblocks(0xFFFFFFFF, buf)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OverflowError: overflow converting long int to machine word
>>> bd.readblocks(0, buf)
True
>>> bd.readblocks(0x7FFFFFFF, buf)
False
>>> bd.readblocks(0, buf)
True
>>> bd.readblocks(0x8FFFFFFF, buf)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OverflowError: overflow converting long int to machine word
>>> bd.readblocks(0, buf)
True
>>> bd.readblocks(0x1000000, buf)
False
>>> bd.readblocks(-20, buf)
False
>>> bd.readblocks(0, buf)
True
>>> bd.readblocks(-2, buf)
False
>>> bd.readblocks(0, buf)
True
>>> bd.readblocks(-1, buf)
(hangs)
```

It seems like something peculiar to -1

projectgus, 2025-02-18:
> It seems like something peculiar to -1

Figured this out, see the linked PR for more explanation.

PR #16703 (2025-02-25): stm32: Fix hang when negative block_num passed to sdcard.
### Summary

* The ST HAL does a [bounds check on SD read operations](https://github.com/micropython/stm32lib/blob/vendor/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_sd.c#L1092), but there's an unsigned integer overflow bug if `(BlockAdd + NumberOfBlocks)` overflows. It seems like some SD cards then reject the out-of-bounds block number, but some become unresponsive. Closes #10622.
* The operation actually will time out (although the card won't respond to subsequent commands.) The timeout was set to 60 seconds and for some reason (unclear) m [... cut]
closed 2025-02-25 by dpgeorge