micropython/micropython #10625: Error MBEDTLS_ERR_SSL_CONN_EOF on api request with Pico W and display

jonbhanson, 2023-01-30:
I am trying to access [my test endpoint](https://us-central1-boiler-monitor-c6c14.cloudfunctions.net/helloWorld) using a Pico W with a [Pimoroni Display Pack 2.0](https://shop.pimoroni.com/products/pico-display-pack-2-0?variant=39374122582099), using the their [pimoroni-picow-v1.19.12-micropython.uf2 ](https://github.com/pimoroni/pimoroni-pico/releases).  (I mention this only because it decreases the memory which may be related to the problem).

I've been having a lot of trouble getting the error `(-29312, 'MBEDTLS_ERR_SSL_CONN_EOF')`, so I've trimmed my code down to the basics to try to get to the bottom of the trouble.  Here's my basic code:

```py
import time
import network
import urequests
import gc
import micropython

from pimoroni import RGBLED
from picographics import PicoGraphics, DISPLAY_PICO_DISPLAY_2,

# set up the hardware
display = PicoGraphics(display=DISPLAY_PICO_DISPLAY_2, rotate=0)

print("starting...")
def network_connect():

    # Set your wifi ssid and password here
    WIFI_SSID = const('wifi')
    WIFI_PASSWORD = const('')

    # Start connection
    wlan = network.WLAN(network.STA_IF)
    wlan.active(True)
    wlan.connect(WIFI_SSID, WIFI_PASSWORD)

    # Wait for connect success or failure
    max_wait = 20
    while max_wait > 0:
        if wlan.status() < 0 or wlan.status() >= 3:
            break
        max_wait -= 1
        print('waiting for connection...')
        time.sleep(1)

    # Handle connection error
    if wlan.status() != 3:
        raise RuntimeError('wifi connection failed %d' % wlan.status())
    else:
        print('connected')
        status = wlan.ifconfig()
        print( 'ip = ' + status[0] )
network_connect()

def poll_server():
    try:
        r = urequests.get("https://us-central1-boiler-monitor-c6c14.cloudfunctions.net/helloWorld")
        print (r.text)
    except Exception as e:
        print(e) 

while True:
    print ("=============")
    print(micropython.mem_info())
    poll_server()
    time.sleep(5)
```

When I run this, here's the output:

```
starting...
connected
ip = 10.0.0.78
=============
stack: 548 out of 7936
GC: total: 166016, used: 85616, free: 80400
 No. of 1-blocks: 132, 2-blocks: 30, max blk sz: 4800, max free sz: 5022
None
Hello from Firebase!
=============
stack: 548 out of 7936
GC: total: 166016, used: 90736, free: 75280
 No. of 1-blocks: 228, 2-blocks: 42, max blk sz: 4800, max free sz: 1740
None
Hello from Firebase!
=============
stack: 548 out of 7936
GC: total: 166016, used: 95856, free: 70160
 No. of 1-blocks: 324, 2-blocks: 54, max blk sz: 4800, max free sz: 1598
None
Hello from Firebase!
=============
stack: 548 out of 7936
GC: total: 166016, used: 100976, free: 65040
 No. of 1-blocks: 420, 2-blocks: 66, max blk sz: 4800, max free sz: 1598
None
Hello from Firebase!
=============
stack: 548 out of 7936
GC: total: 166016, used: 106096, free: 59920
 No. of 1-blocks: 516, 2-blocks: 78, max blk sz: 4800, max free sz: 1598
None
Hello from Firebase!
=============
stack: 548 out of 7936
GC: total: 166016, used: 111216, free: 54800
 No. of 1-blocks: 612, 2-blocks: 90, max blk sz: 4800, max free sz: 1598
None
(-29312, 'MBEDTLS_ERR_SSL_CONN_EOF')
=============
stack: 548 out of 7936
GC: total: 166016, used: 86784, free: 79232
 No. of 1-blocks: 104, 2-blocks: 25, max blk sz: 4800, max free sz: 3985
None
(-29312, 'MBEDTLS_ERR_SSL_CONN_EOF')
=============
stack: 548 out of 7936
GC: total: 166016, used: 89200, free: 76816
 No. of 1-blocks: 120, 2-blocks: 26, max blk sz: 4800, max free sz: 3985
None
Hello from Firebase!
=============
stack: 548 out of 7936
GC: total: 166016, used: 94320, free: 71696
 No. of 1-blocks: 216, 2-blocks: 38, max blk sz: 4800, max free sz: 2353
None
Hello from Firebase!
=============
stack: 548 out of 7936
GC: total: 166016, used: 99440, free: 66576
 No. of 1-blocks: 312, 2-blocks: 50, max blk sz: 4800, max free sz: 2353
None
Hello from Firebase!
=============
stack: 548 out of 7936
GC: total: 166016, used: 104560, free: 61456
 No. of 1-blocks: 408, 2-blocks: 62, max blk sz: 4800, max free sz: 2352
None
Hello from Firebase!
=============
stack: 548 out of 7936
GC: total: 166016, used: 109680, free: 56336
 No. of 1-blocks: 504, 2-blocks: 74, max blk sz: 4800, max free sz: 2234
None
Hello from Firebase!
=============
stack: 548 out of 7936
GC: total: 166016, used: 114800, free: 51216
 No. of 1-blocks: 600, 2-blocks: 86, max blk sz: 4800, max free sz: 1324
None
(-29312, 'MBEDTLS_ERR_SSL_CONN_EOF')
=============
stack: 548 out of 7936
GC: total: 166016, used: 86784, free: 79232
 No. of 1-blocks: 104, 2-blocks: 25, max blk sz: 4800, max free sz: 3631
None
(-29312, 'MBEDTLS_ERR_SSL_CONN_EOF')
=============
stack: 548 out of 7936
GC: total: 166016, used: 89200, free: 76816
 No. of 1-blocks: 120, 2-blocks: 26, max blk sz: 4800, max free sz: 3631
None
Hello from Firebase!
=============
stack: 548 out of 7936
GC: total: 166016, used: 94320, free: 71696
 No. of 1-blocks: 216, 2-blocks: 38, max blk sz: 4800, max free sz: 1997
None
Hello from Firebase!
=============
stack: 548 out of 7936
GC: total: 166016, used: 99440, free: 66576
 No. of 1-blocks: 312, 2-blocks: 50, max blk sz: 4800, max free sz: 1997
None
Hello from Firebase!
```

As can be seen from the output, the `MBEDTLS_ERR_SSL_CONN_EOF` error is hindering the code from reliably reading the api data.

MATTYGILO, 2023-07-17:
Have your found a fix for this?

chanukab7, 2023-11-18:
same, found a fix yet? @jonbhanson @MATTYGILO

jonbhanson, 2023-11-20:
I have not able to find a resolution for this.

MATTYGILO, 2023-11-20:
@chanukab7 i haven't found a fixed either I just added a minimum time between calls and retirees on errors

chanukab7, 2023-11-21:
**ap_config.py**

```
# ap_config.py

import network, socket, time

def connect_to_wifi(ssid: str, password: str):
    """ Connect to a wifi network using the ssid and password provided
    ssid: wifi network name
    password: wifi network password """

    wlan = network.WLAN(network.STA_IF)
    
    wlan.active(False)
    time.sleep(1)
    wlan.active(True)

    wlan.disconnect()

    if not wlan.isconnected():
        print('connecting to network...')
        wlan.connect(ssid, password)
        while not wlan.isconnected():
            pass
            
    print("WIFI Connected : ", wlan.ifconfig()[0])
    return wlan, wlan.ifconfig()[0]

def start_ap(name: str, passwd: str):
    """ Start an access point with the essid and password provided
    essid: access point name
    password: access point password """

    ap = network.WLAN(network.AP_IF)
    ap.active(False)
    time.sleep(1)
    ap.active(True)
    ap.config(essid=name, password=passwd)
    time.sleep(1)

    print("Access Point: ", ap.ifconfig()[0])
    return ap, ap.ifconfig()[0]

def start_socket_tcp():
    """ Start a TCP socket on the port provided.
    returns the socket and the port
    ip: ip address to start the socket on
    port: port number to start the socket on
    max_connections: maximum number of connections to accept """

    sckt = socket.socket(socket.AF_INET, socket.SOCK_STREAM)

    ai = socket.getaddrinfo("0.0.0.0", 8443)
    print("Bind address info:", ai)
    addr = ai[0][-1]
    sckt.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)

    sckt.bind(addr)
    print('listening on', 8443)
    return sckt, 8443
```

**main.py**

```
# main.py
from ap_config import *
import time
import ssl
import gc

CONTENT = b"""\
HTTP/2 200 OK
Content-Type: text/html
Access-Control-Allow-Origin: *

%s
"""

with open('data.csv', 'rb') as f:
    data = f.read()

def load_certificate_and_key():
    with open('server.key', 'rb') as f:
        key = f.read()
    with open('server.crt', 'rb') as f:
        cert = f.read()
    return key, cert

def handle_request(conn):
    try:
        req = conn.readline(10)
        if req == b"" or req == b"\r\n":
            pass
        elif "GET /req" in req[:10]:
            conn.write(CONTENT % data)
            print(CONTENT % data)
        else:
            pass
           
    except Exception as e:
        print("Exception serving request:", e)

    conn.close()
    print()

# WiFi connection / socket initialization
connect_to_wifi("developer", "")
ap_socket, _ = start_socket_tcp()
ap_socket.settimeout(0.1)

# Load the key and the certificate
key, cert = load_certificate_and_key()

while True:
    try:
        ap_conn, addr = ap_socket.accept()
        print('Got a connection from %s via the local server' % str(addr))
        conn = ssl.wrap_socket(ap_conn, key=key, cert=cert, server_side=True)
        handle_request(conn)
    except Exception as e:
        pass
    gc.collect()
```

Well this worked for me. takes about 8 seconds to fully establish a connection and retrieve content in a browser though.