micropython/micropython #10634: ESP32 SDCard OSError: 16

engineertrooper, 2023-01-31:
Hi Community,

in the discussion the problem is explained. Also one solution was discussed there already. 
Let me know if I can test a new idea to locate the problem.

Thank you!

--------

### Discussed in https://github.com/orgs/micropython/discussions/10581

<div type='discussions-op-text'>

<sup>Originally posted by **engineertrooper** January 24, 2023</sup>
Hi Community,

**Problem:**
I get an OSError: 16 if I use the example code [V1.19.1](https://docs.micropython.org/en/v1.19.1/library/machine.SDCard.html#machine-sdcard) for the SDCard Class.

**My Device:**
- ESP32 Dev Kit C V2 or also sometimes named ESP32 NodeMCU Module
- with Micropython: v1.19.1 -> 2022-06-18
- Pins: sck=18, miso=19, mosi=23, cs=5 and 3,3V supply

**Pre-Analyse**
An OSError: 16 means a device or ressource is busy. Because of that I commented out all code with exception the following lines:

```python

import os
import machine

sd = machine.SDCard(slot=2)
os.mount(sd, "/sd")  # mount

os.listdir('/sd')    # list directory contents

```

Error-Msg (always the same, it doesn't matter if other code is commented out or not):

```

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4540
ho 0 tail 12 room 4
load:0x40078000,len:12344
ho 0 tail 12 room 4
load:0x40080400,len:4124
entry 0x40080680
Traceback (most recent call last):
  File "main.py", line 62, in <module>
OSError: 16
MicroPython v1.19.1 on 2022-06-18; ESP32 module with ESP32
Type "help()" for more information.
>>> 

```

**Looking into the world wide web, github issues and other discussions**

While I was reading I got the feeling that it has not something to do with the board directly. It looks like to me that the error is inside the ESP-IDF or there was a change in Micropython that was making that error existing?

After reading some Issues, Documentation, Discussions and Tutorials I marked down the following links:

- [Class SDCard](https://docs.micropython.org/en/v1.19.1/library/machine.SDCard.html#machine-sdcard)
- [Class SPI](https://docs.micropython.org/en/v1.9.1/pyboard/library/machine.SPI.html)
- [MicroPython Forum](https://forum.micropython.org/viewtopic.php?t=12007)
- [#8949](https://github.com/micropython/micropython/issues/8949)
- [#8290](https://github.com/micropython/micropython/issues/8290)
- [#9925](https://github.com/micropython/micropython/discussions/9935)
- [#429 micropython-lib](https://github.com/micropython/micropython-lib/issues/429)
- [#8940](https://github.com/micropython/micropython/issues/8940)

**Note: tutorials excluded, because old solutions with a [blob](https://github.com/micropython/micropython/blob/master/drivers/sdcard/sdcard.py) from @dpgeorge was used. But I found something similar of the blob [here](https://github.com/micropython/micropython-infineon/blob/master/drivers/sdcard/sdcard.py).*

***Issue 429 and 8940 maybe not related to this question, but I'm not very fimilar to ESP-IDF, so added them just in case.*

**SD-Card Frequenz**

One try was to set a frequenz. -> ``` freq=20000000 ``` as parameter inside ```SDCard(...)```
I played with diffrent frequenzes between 100KHz and 20MHz, but didn't work. (See [#8290 comment](https://github.com/micropython/micropython/issues/8290#issuecomment-1054208392))

*Note: I'm using a 32GB SanDisk Ultra MicroSdCard.*

**Question:**

Any Idea how this **OSError: 16** is coming up?
Or Do I oversee something simple here?

With these Information I wanted to ask here first, before opening a new issue.

 </div>

robert-hh, 2023-02-01:
According to the referred discussion this seems to be a problem in the set-up of the board & SD-card adapter,

engineertrooper, 2023-02-01:
> According to the referred discussion this seems to be a problem in the set-up of the board & SD-card adapter

The board is broken or damaged... Don't know, but with a TTGO-Koala Esp32 Board it works like a charm.

This issue can be closed -> [answer](https://github.com/orgs/micropython/discussions/10581#discussioncomment-4843230) found

Thank you for your help and idea!