micropython/micropython #10644: Can't run minimal port

Maxi35, 2023-02-02:
I'm trying to get the "Minimal" port to work on my wsl but after building it. When I run it, it cannot execute any commands since it gives memoryError:
```
>>> a=2
GC: total: 1920, used: 1824, free: 96
 No. of 1-blocks: 6, 2-blocks: 1, max blk sz: 32, max free sz: 3
GC: total: 1920, used: 1920, free: 0
 No. of 1-blocks: 7, 2-blocks: 2, max blk sz: 32, max free sz: 0
GC: total: 1920, used: 1920, free: 0
 No. of 1-blocks: 7, 2-blocks: 2, max blk sz: 32, max free sz: 0
MemoryError:
>>>
Exit status: 0
```

Info about how it was build:
gcc version:
`gcc (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0`

Pulled a fresh micropython repo today, navigated to the minimal port and build using `make` then run with `make run`

REPL version information: `MicroPython v1.19.1-850-gfe2a8332f on 2023-02-02; minimal with unknown-cpu`

Maxi35, 2023-02-02:
Yea many thanks!

manobendro, 2023-02-02:
You may close the issue.

closed 2023-02-02 by Maxi35