micropython/micropython #10648: Extending MicroPython in C

hdpklm, 2023-02-02:
Hello

I try to build the native example from this link: https://docs.micropython.org/en/latest/develop/natmod.html#minimal-example
but I get this error
```sh
$ make ARCH=xtensa
process_begin: CreateProcess(NULL, /bin/mkdir -p build, ...) failed.
make (e=2): El sistema no puede encontrar el archivo especificado.  
make: *** [build] Error 2
```

I change MPY_DIR to the direction of micropython folder.

os: windows 10
python: 3.11.1
GNU Make 3.81

I think the path `/bin/mkdir` is for linux not for windows.

any help please

manobendro, 2023-02-02:
You may use WSL, Google cloud shell, or a GitHub codespace for a free Linux machine. If you try to build for an ESP32 chip then follow the port specification.

hdpklm, 2023-02-02:
Ok I try to buind in a docker machine and work fine, but its easy to work in the windows because docker has no visual ide.

what are the port specification ???

closed 2023-02-06 by hdpklm