micropython/micropython #10683: Wrong results when rounding some numbers

pneumokoniose, 2023-02-06:
Hi,

I have strange errors when rounding some numbers:

```
>>> round(73.84343, 1)
73.80001 
>>> round(74.84343, 1)
74.8
```

Other example:

```
>>> round(71.30001, 1)
71.30001 
>>> round(72.30001, 1)
72.3
`````

I work on a Raspberry Pi Pico and Micropython 1.19.1.

dlech, 2023-02-06:
Many floating point values don't have an exact representation, so rounding issues like this are expected. If you just want to print the number with one decimal point, you should use string formatting rather than `round()`. 

```python
>>> "%.1f" % 73.84343
'73.8'
>>> "{:.1f}".format(73.84343)
'73.8'
>>> "{73.84343:.1f}"
'73.8'
```

pneumokoniose, 2023-02-06:
> Many floating point values don't have an exact representation, so rounding issues like this are expected. If you just want to print the number with one decimal point, you should use string formatting rather than `round()`.
> 
> ```python
> >>> "%.1f" % 73.84343
> '73.8'
> >>> "{:.1f}".format(73.84343)
> '73.8'
> >>> f"{73.84343:.1f}"
> '73.8'
> ```

Thank you! I was under the assumption that I say this explicitly with round(n, 1). What is the reason why I get the result as expected under Python?

Python:
`>>> round(71.30001, 1)
71.3`
```
MicroPython:
>>> round(71.30001, 1)
71.30001 
`

I hope this is not a stupid question.

dlech, 2023-02-06:
CPython uses `double` for floating point numbers (64-bit) but most MicroPython ports on embedded systems use `float` (32-bit) instead. Since there are fewer bits available to store the number there are fewer numbers with an exact representation in MicroPython.

pneumokoniose, 2023-02-06:
> CPython uses `double` for floating point numbers (64-bit) but most MicroPython ports on embedded systems use `float` (32-bit) instead. Since there are fewer bits available to store the number there are fewer numbers with an exact representation in MicroPython.

Interesting, thank you!

closed 2023-02-06 by pneumokoniose