micropython/micropython #10690: rp2: micropython.schedule() thread safety

peterhinch, 2023-02-07:
The canonical use for `micropython.schedule()` is to enable a hard ISR to schedule a "soft" ISR which has GIL protection. For this to occur the "soft" code must run on the same core (0) as the main program.

In the following script the main program and the hard ISR run on core 0 while `soft_isr` runs on core 1. If `soft_isr` were to access shared data there would be a risk of corruption, for example if it accessed a global while the globals dict was being re-hashed.

The problem is that user code is running on another core under circumstances in which this would not be expected.
```python
import micropython
import _thread
import time
from machine import Pin, mem32

pout = Pin(16, Pin.OUT)  # Link these pins
ipin = Pin(17, Pin.IN)

def pcore(s):
    print(f'{s} CPU core: {mem32[0xd0000000]} thread: {_thread.get_ident()}')

def isr(_):
    print(mem32[0xd0000000])
    micropython.schedule(soft_isr, None)

ipin.irq(handler=isr, hard=True)

def soft_isr(_):  # Scheduled by hard ISR
    pcore('soft')

def busy():
    while True:
        time.sleep(1)
        pcore('thread')

_thread.start_new_thread(busy, ())
pcore('main')
while True:
    pout(not pout())
    time.sleep_ms(200)
```
Outcome:
```
<irq>
main CPU core: 0 thread: 537095120
0
soft CPU core: 1 thread: 536912624
0
soft CPU core: 1 thread: 536912624
0
soft CPU core: 1 thread: 536912624
0
soft CPU core: 1 thread: 536912624
0
soft CPU core: 1 thread: 536912624
thread CPU core: 1 thread: 536912624
0
soft CPU core: 1 thread: 536912624
0
```
This arose from discussion https://github.com/micropython/micropython/discussions/10638

dlech, 2023-02-07:
A while ago, I made some arguments in favor of only running scheduled callbacks on the main thread (mainly with the unix port in mind at the time rather than RP2) at https://github.com/micropython/micropython/pull/8838 but never pursued it any farther. Seems worth repeating here since it would solve more problems than just the specific one that Peter identified.

> For example, could we make it so that [the scheduler only runs on the main thread](https://github.com/pybricks/micropython/commit/b196c783c081c5444749995df789d1620fa90814)? This is inline with how CPython does things (i.e. [Py_AddPendingCall](https://docs.python.org/3/c-api/init.html#c.Py_AddPendingCall)). And it seems to me like it would be beneficial at least in the case where there is an unhandled exception in a scheduled callback. If the scheduler can run on any thread, and there is an unhandled exception in a scheduled callback, then it will just crash the random thread where it ran and the rest of the program will keep running in a bad state. But if all scheduled callbacks only run on the main thread and there is an unhandled exception, then the entire program crashes instead of continuing to run in a bad state. This seems like a more desirable behavior.

rkompass, 2023-02-07:
I would prefer both determinism and flexibility, so my question: Would it be possible to have a third optional argument in `micropython.schedule()` that determines the core? With a default setting to core 0? I can imagine situations where the second core is not busy and (if it is there, like on RP2) can be utilized to serve the scheduled task faster. If selected as an non-default setting this should include awareness to the multiple threading pitfalls.
On the other hand this would introduce threading without `import _thread`.
... I should test our examples without this import to see if this is already the case. My current understanding is that without `import _thread` the second core is not used at all. Is this correct?

peterhinch, 2023-02-08:
>  My current understanding is that without import _thread the second core is not used at all. Is this correct? Tested: Yes, all scheduled tasks then happen on core 0.

My testing confirms this and is the reason my test case uses threading.

redhead-p, 2023-02-11:
As noted in #10638 the same indeterminacy regarding core is evidenced in callbacks initiated by `machine.timer()` objects and, I suspect, in 'soft' ISRs when linked to pins and state machines.  The solution adopted needs to work for these implicit uses of `_thread.schedule()` or whatever the underlying mechanism is in these cases.

In terms of an immediate bug fix - running all 'soft' ISRs on core 0 as @dlech suggests would at least make the situation deterministic. I'd assume here, that at the moment, only the NVIC on core 0 is used for the vectoring of 'hard' ISRs even when `_thread` is imported. I stand to be corrected by the experts here @jimmo.

It seems sensible that transfer of control and synchronisation of operations between the two cores should only use the functions explicitly provided within the `_thread` library and there should be no 'back doors'.  Therefore a transfer from 'hard' to 'soft' interruptible context should result in running on the same core as the originating hard ISR.

If flexibility is required on which core the soft ISR runs, then we should be able to specific which core's NVIC is used to vector the interrupt to the 'hard' ISR rather than modify the `_thread.schedule()` API. I'd view such changes as an enhancement rather than a bug fix.

peterhinch, 2023-02-12:
> It seems sensible that transfer of control and synchronisation of operations between the two cores should only use the functions explicitly provided within the _thread library and there should be no 'back doors'.

I agree completely. The user needs to be entirely clear when a process will run on another thread/core so that thread safe precautions can be taken.