micropython/micropython #10692: WASM port: passes `-sASYNICIFY` but doesn't use it

hoodmane, 2023-02-08:
If I remove `-sASYNCIFY` the test suite runs approximately 3x faster (on my local machine, an average of 52.6 vs 18.4s over three trials each).

I can't find that any `ASYNC_JS` functions are ever called (`emscripten_sleep`, `emscripten_wget`, etc) and when you use `ccall` you never pass `{async: true}`, so if an `ASYNC_JS` function were called it would cause an assertion failure. So I think all `-sASYNCIFY` is doing is making calling function pointers way more expensive for no gain.

@ntoll

hoodmane, 2023-02-08:
Also, `--memory-init-file 0` doesn't do anything except when targeting asmjs (instead of wasm), so that should probably be removed as a linker argument.

dpgeorge, 2023-02-08:
I was under the impression that `-s ASYNCIFY` was needed for `emscripten_scan_stack()` and `emscripten_scan_registers()` to work correctly.

dpgeorge, 2023-02-08:
See #7449.

hoodmane, 2023-02-08:
It looks to me like it is needed for `emscripten_scan_registers` but not `emscripten_scan_stack`. Is there test coverage for this? If an import that needed asyncify is called inside of a `ccall` and `{async: true}` was not passed, it should fail with an assertion error. So the fact that we don't see that assertion error suggests that `emscripten_scan_registers` is not actually being called. Unless I'm misunderstanding something.

dpgeorge, 2023-02-08:
> Is there test coverage for this?

There is... but it's not being run.  There are tests in `tests/stress/` that will exercise the GC but they aren't run during CI on the webassembly port.

I checked, and simply running `gc.collect()` will fail with a (Javascript/WASM) assertion error.

So we'll need to add the async option to all calls, at least `mp_js_do_str` and `mp_js_process_char`.

dpgeorge, 2023-02-08:
Side note: for the case of pyscript, it may be possible to work around the need to enable ASYNCIFY (because having it disabled will give a big performance boost, and generate smaller WASM binaries).

The idea would be to disable the GC during execution of Python code, and only run it on entry and/or exit of Python code.  That way you don't need to scan the stack or registers because the only root pointers are those in the known root pointer set (ie there are no root pointers on the C stack or in registers).

Then to handle the case that the running Python code runs out of memory (there is no GC so it cannot reclaim any), we can expand the MicroPython heap by allocating more from Javascript.

hoodmane, 2023-02-08:
>  simply running `gc.collect()` will fail with a (Javascript/WASM) assertion error.

Okay this is reassuring.

> it's not being run.

For testing, the vast majority of the run time is repeated startups. Since you don't use dynamic linking or pthreads or anything messy, you should be able to take a memory snapshot with 
```js
let snapshot = Module.HEAP8.slice();
```
before running any tests. Then after each test, run 
```js
Module.HEAP8.set(snapshot);
```
to restore the memory before each test. I think the test execution time will be reduced to very very small.

The easiest way to do this would be with a pytest fixture but the tests don't use pytest. It's probably possible to make an appropriate node process server instead but it's a bit more work.

ntoll, 2023-02-08:
So, on my TODO list was to introduce the two of you... but I see you're already chatting. This is great to see, and I'm confident you'll both have a LOT of fun collaborating and sharing ideas. :tada:

hoodmane, 2023-02-08:
Actually the FS state will not be restored by this since the FS state is stored in JavaScript. I'm not sure if there is an easy way to reset FS state.

dpgeorge, 2023-06-27:
Fixed by 813d559bc098eeaa1c6e0fa1deff92e666c0b458

commit b2ad7e23 (2023-06-27): webassembly: Make mp_js_do_str asynchronous.

This fixes a bug where `gc.collect()` would crash due to
emscripten_scan_stack being called synchronously within mp_js_do_str.  The
fix is to make mp_js_do_str asynchronous.

Fixes #10692.
closed 2023-06-27 by dpgeorge