micropython/micropython #10711: ubluetooth missing from GENERIC_S3_SPIRAM-20220618-v1.19.1.bin

AeroSynth, 2023-02-10:
### Discussed in https://github.com/orgs/micropython/discussions/10710

<div type='discussions-op-text'>

<sup>Originally posted by **AeroSynth** February 10, 2023</sup>
I have two ESP32-S3 Wroom DevKitC boards.
One has 2K SPIRAM 8M Flash and the other only has the 8M Flash.
When I started testing GENERIC_S3_SPIRAM-20220618-v1.19.1.bin, the ubluetooth library seems to be missing.  It's not listed when you type help(modules).
However, the other image GENERIC_S3-20220618-v1.19.1.bin does include ubluetooth and it seems to function running the BLE examples.
I also noted that the SPIRAM image is smaller than the other one suggesting it may be missing some bits.</div>

mattytrentini, 2023-02-11:
Ah, I see - back in November there was an [update to GENERIC_S3_SPIRAM](https://github.com/micropython/micropython/commit/b705732d5f385721f5c50d9735252bfe10f180ec) that enabled bluetooth; it had previously been inadvertently disabled. This includes being disabled during the 1.19 release.

If you can pick up a [nightly build of GENERIC_S3_SPIRAM](https://micropython.org/download/GENERIC_S3_SPIRAM/) it should be fine.

AeroSynth, 2023-02-11:
Yep.  That fixed it.
Thanks!

mattytrentini, 2023-02-11:
Glad to hear that resolved the issue.

closed 2023-02-11 by mattytrentini