micropython/micropython #10719: docs: readme of esp32 port says it doesn't automatically connect wifi on boot

dequis, 2023-02-11:
https://github.com/micropython/micropython/tree/master/ports/esp32#configuring-the-wifi-and-using-the-board says:

>The ESP32 port is designed to be (almost) equivalent to the ESP8266 in terms of the modules and user-facing API. There are some small differences, notably that the ESP32 does not automatically connect to the last access point when booting up. 

That's... not true anymore, is it? Seems to contradict what the quickref says: https://docs.micropython.org/en/latest/esp32/quickref.html#wlan

jimmo, 2023-02-15:
> That's... not true anymore, is it? Seems to contradict what the quickref says: https://docs.micropython.org/en/latest/esp32/quickref.html#wlan

That is still true. _Only_ the ESP8266 connects automatically at boot (which we unfortunately cannot change without breaking backwards compatibility).

I don't see the contradiction in the quickref. @dequis can you highlight specifically the bit that you think is wrong?

dequis, 2023-02-15:
Huh! Weird.

>I don't see the contradiction in the quickref. @dequis can you highlight specifically the bit that you think is wrong?

"After a call to wlan.connect(), the device will by default retry to connect **forever**, even when the authentication failed or no AP is in range" (my emphasis)

jonnor, 2024-09-01:
The documentation referred above seems acceptable to me.

projectgus, 2025-02-04:
Closing as I believe that's correct: the esp32 only connects wifi on boot if you've added a `boot.py` file to the filesystem that runs code to connect wifi (in which case it will have whatever connection/reconnect behaviour that's implementated in that code.) Otherwise, it doesn't enable wifi on boot at all.

closed 2025-02-04 by projectgus