micropython/micropython #10722: `ussl` not in board `ARDUINO_NANO_RP2040_CONNECT` in port `rp2`

mm12, 2023-02-11:
Firmware: ARDUINO_NANO_RP2040_CONNECT-20220618-v1.19.1.uf2
Version: (sysname='rp2', nodename='rp2', release='1.19.1', version='v1.19.1 on 2022-06-18 (GNU 11.2.0 MinSizeRel)', machine='Arduino Nano RP2040 Connect with RP2040')
Port: `rp2`
Board: `ARDUINO_NANO_RP2040_CONNECT`
---
`mpconfigboard.cmake` and `mpconfigboard.h` do not include an enable of `ussl`, and as a result, an input error occurs when attempting to use networking, such as `urequests`

robert-hh, 2023-02-11:
I see it on my board with a more recent firmware. So if you load the actual nightly build, you will get ussl.

mm12, 2023-02-11:
Solved, just didn't see it in the repo.

closed 2023-02-11 by mm12