micropython/micropython #10751: codeformat.py bug

mbedNoobNinja, 2023-02-14:
## Description:

When codeformat.py formats `*.h` or `*.c` files (#10595) , it reduces file content by deleting not only formating symbols and whitespaces but also the **code** being formated! 

[before_codeformat.txt](https://github.com/micropython/micropython/files/10731651/before_codeformat.txt)
[after_codeformat.txt](https://github.com/micropython/micropython/files/10731653/after_codeformat.txt)

and while working on `before_codeformat.h` **codformat.py** gives the folowing error:

```
Parsing: ..\ports\renesas-ra\boards\VK_RA4W1\ra_cfg\fsp_cfg\bsp\bsp_cfg.h as language C
Traceback (most recent call last):
   File "D:\micropython\tools\codeformat.py", line 218, in <module>
     main()
   File "D:\micropython\tools\codeformat.py", line 205, in main
     fixup_c(file)
   File "D:\micropython\tools\codeformat.py", line 136, in fixup_c
     if dedent_stack[-1] >= 0:
        ~~~~~~~~~~~~^^^^
IndexError: list index out of range
```

## Environment:

| OS | Win 10 (22H2) |
| :---: | :---: |

| Uncrustify | 0.72.0_f |
| :---: | :---: |

| Python | 3.11.1 |
| :---: | :---: |

| Package | Version |
| :---: | :---: |
| black | 22.12.0 |
| click | 8.1.3 |
| colorama | 0.4.6 |
| mypy-extensions | 0.4.3 |
| pathspec | 0.11.0 |
| pip | 23.0 |
| platformdirs | 2.6.2 |
| pyserial | 3.5 |
| setuptools | 65.5.0 |

dpgeorge, 2023-02-14:
`codeformat.py` is confused by the uneven indenting of the C preprocessor directives in `before_codeformat.h`.  The simplest way to fix this is to just de-dent all lines in that header file.

mbedNoobNinja, 2023-02-14:
Yes, I noticed the iidenting should be even, but the problem is that those header files are auto generated and they are unevenly idented every time I hit build in the IDE, but thats not a pain to die from, I just have to watch out when I commit.

dlech, 2023-02-14:
If you can modify the code generator or add an extra step in the code generation pipline, you can have it add `// *FORMAT-OFF*` to the problematic code. You can also tell `codeformat.py` to exclude specific files or directories.

jonnor, 2024-09-01:
Is there something to fix here?