micropython/micropython #10766: REPL parse error of valid bytes data.

antirez, 2023-02-16:
To reproduce, just parse the following in the REPL:

```Python
b'\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\xaa'
```

The result will be a syntax error, even if the string is valid and can be parsed by Python3.

robert-hh, 2023-02-16:
Note here. The string is parsed identically on CPython 3.9.2 and  `MicroPython v1.19.1-869-gde1f1dd16 on 2023-02-16`. The result is in both cases:

b'\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\xaa'

antirez, 2023-02-16:
Very odd. I'm using this image:

https://micropython.org/download/LILYGO_TTGO_LORA32/

robert-hh, 2023-02-16:
Try the most recent nightly build. There have been changes.

antirez, 2023-02-16:
[image]

antirez, 2023-02-16:
I'll try the most recent build and report back. Thank you for your kind help. Appreciated.

antirez, 2023-02-16:
After upgrading:

```Python
MPY: soft reboot
MicroPython v1.19.1-869-gde1f1dd16 on 2023-02-16; LILYGO TTGO LoRa32 with ESP32
Type "help()" for more information.
>>> b'\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a\x5cx5a'
b'\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a\\x5a'
>>> $zx5a\x5cx5a\x5cx5a\x5cx5a\xaa'
Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: invalid syntax
```

robert-hh, 2023-02-16:
No problem here with an ESP32. It looks as if on your side the input is split into two parts. How do you paste it into the board?
Do you use Thonny? That may be the culprit. Use some other terminal tool like mpremote, Putty, screen, or anything else that does not trick you out.

antirez, 2023-02-16:
About the memory corruption, as you can see it shows `$zx` that is not even in the original input. I believe the problem could be in the handling of escapes.

antirez, 2023-02-16:
Oh, I forgot to say an important thing: this also happens when writing from a C program to the USB device, with the REPL set in RAW MODE.

robert-hh, 2023-02-16:
The screen dumps you show do not look like raw mode. How do you set REPL to raw mode (with Ctrl-A) and how do you write?

antirez, 2023-02-16:
Sorry I was not clear: I discovered the issue in raw mode, using a program that sends CTRL+A, then writes (write(2) system call) the same string and so forth. After debugging the program, I found that the same issue also happen without raw mode.

This is another instance of the same bug (occurs when the lines are long, apparently):
```
bytes([97,117,115,101,32,108,105,99,101,110,115,101,46,10,35,32,83,101,101,32,116,104,101,32,76,73,67,69,78,83,69,32,102,105,108,101,32,102,111,114,32,109,111,114,101,32,105,110,102,111,114,109,97,116,105,111,110,10,10,105,109,112,111,114,116,32,109,97,99,104,105,110,101,44,32,115,115,100,49,51,48,54,44,32,115,120,49,50,55,54,44,32,116,105,109,101,44,32,117,114,97,110,100,111,109,44,32,103,99,10,102,114,111,109,32,109,97,99,104,105,110,101,32,105,109,112,111,114])
```

And this is the session using raw mode:

```
MicroPython v1.19.1-869-gde1f1dd16 on 2023-02-16; LILYGO TTGO LoRa32 with ESP32
Type "help()" for more information.
>>> 
raw REPL; CTRL-B to exit
>OKTraceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: bytes value out of range
```

And, with the normal REPL mode:

Cut & paste of my clipboard before pasting it into MicroPython with screen:

```
bytes([97,117,115,101,32,108,105,99,101,110,115,101,46,10,35,32,83,101,101,32,116,104,101,32,76,73,67,69,78,83,69,32,102,105,108,101,32,102,111,114,32,109,111,114,101,32,105,110,102,111,114,109,97,116,105,111,110,10,10,105,109,112,111,114,116,32,109,97,99,104,105,110,101,44,32,115,115,100,49,51,48,54,44,32,115,120,49,50,55,54,44,32,116,105,109,101,44,32,117,114,97,110,100,111,109,44,32,103,99,10,102,114,111,109,32,109,97,99,104,105,110,101,32,105,109,112,111,114])
```

But as soon as I open screen and paste the same thing, the echo produced by MicroPython is already corrupted, and the output now shows '444' that was non existing before:

```
>>> bytes([97,117,115,101,32,108,105,99,101,110,115,101,46,10,35,32,83,101,101,32,116,104,101,32,76,73,67,69,78,83,69,32,102,105,108,101,32,102,111,114,32,109,111,114,101,32,105,110,102,111,114,109,97,116,105,111,110,10,10,105,109,112,111,114,116,32,109,97,99,104,105,110,101,44,32,115,115,100,49,51,48,54,44,32,115,120,49,50,55,54,44,32,116,105,109,101,444,32,103,99,10,102,114,111,109,34,32,103,99,10,102,114,111,109,32,109,97,99,104,105,110,101,32,105,109,112,111,114])
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: bytes value out of range
```

So what happens is that:

1. I paste a large value.
2. What I see on the screen is the echo produced by MicroPython, as screen disables local echoing.
3. The echo is corrupted, and what it parses is the corrupted version.

antirez, 2023-02-16:
Oops... I'm very sorry, I understand what it is:

I used the wrong serial device from MacOS:

/dev/tty.usbmodem54790294581 instead of /dev/tty.wchusbserial54790294581

Apparently it does strange things with the input! Closing, and thanks for the help.

closed 2023-02-16 by antirez