micropython/micropython #10773: Native module example - attribute not found

guyturner797, 2023-02-17:
Hi all,

I am trying to build the natmod feature example, and have been able to successfully compile it. However, once transferred to a micropython device, it is importable but the attributes are not available. 

I am building on a Ubuntu VM with micropython and the ESP32 toolchain. I'm then running this on an ESP32-S3 (TinyS3 from Unexpected Maker). 

The output of the make command is:
`~/micropython/examples/natmod/features0$ make ARCH=xtensawin
CC features0.c
LINK build/features0.o
arch:         EM_XTENSA
text size:    101
rodata size:  8
bss size:     0
GOT entries:  3
GEN features0.mpy
`

The micropython code I have tested with is:
`import factorial
print(factorial.__name__)
print(dir(factorial))
print(factorial.factorial(12))`

This provides the output:
`factorial
['__class__', '__name__']
Traceback (most recent call last):
  File "<stdin>", line 14, in <module>
AttributeError: 'module' object has no attribute 'factorial'`

Any advice would be greatly appreciated. Thank you

dpgeorge, 2023-02-18:
It could be that the version of the firmware you have on the device doesn't match the version of the repo at which you compile the native module.  They need to match exactly.  Use `sys.version` to get the version of the esp32 firmware, eg "3.4.0; MicroPython v1.19.1-872-g4bd4b6a47 on 2023-02-17".  That should match `git describe` run in the repo.

guyturner797, 2023-02-18:
Thanks for your reply Damien. This solved my issue, I was running v1.19.1 across both esp32 and ubuntu but hadn't appreciated the exact nightly build revision also needed to match.

closed 2023-02-18 by guyturner797