micropython/micropython #10778: f-string debug expressions use str instead of repr

BrianPugh, 2023-02-18:
When using f-string debug expressions, micropython incorrectly uses (I think) `__str__` instead of `__repr__`.

CPython:
```
Python 3.9.9 (main, Dec 31 2021, 11:22:40)
>>> foo = "hello\nworld\n"
>>> print(f"{foo=}")
foo='hello\nworld\n'
>>>
```

MicroPython (commit 4598b89ce9bd171dde9; Fri Feb 3 17:03:31 2023):
```
MicroPython v1.19.1-628-gecd4d5439 on 2022-11-08; darwin [GCC 4.2.1] version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> foo = "hello\nworld\n"
>>> print(f"{foo=}")
foo=hello
world

>>> print(f"foo={repr(foo)}")  # This mimics the output of cpython
foo='hello\nworld\n'
>>>
```

dpgeorge, 2023-02-21:
It looks like you are correct, MicroPython uses `__str__` instead of `__repr__`.

It's possible to fix, but of course the fix would come at a cost.  Not sure if it's worth that cost.

PsuFan, 2024-03-05:
Was there a change to this? I think my functionality changed. I was using __str__ and it was working. I'm currently using esp32 1.21.0 and its now using repr...

```
class Battery():
	def __str__(self):
		return f'str'
	def __repr__(self):
		return f'repr'

>>> b = Battery()
>>> b
repr

stinos, 2024-03-05:
Your example doesn't have anything to do with f-strings though because there's nothing to interpolate, and it behaves the same in CPython i.e. using `__repr__` on the repl is normal behavior?

PsuFan, 2024-03-05:
@dpgeorge 

that’s getting above my pay grade. Maybe they are still related?

dpgeorge, 2024-03-05:
> Maybe they are still related?

No, they are not related.  Your issue is about printing user objects at the REPL.  That behaviour has been the same as CPython from the beginning of MicroPython (about 10 years ago) in this commit: 76d982ef343dcadd35355aed9c568984c850fb7b

So, nothing has changed in that respect.  Printing an object at the REPL has always used `__repr__`.

PsuFan, 2024-03-06:
@dpgeorge  I 100% had my esp32 repl printing out using __ str __. I can try and find the version if you are interested, might have been 1.19.

PsuFan, 2024-03-06:
>>>print(class) uses str but >>>class() uses repr

Its possible that this changed, but I guess I dont care that much to find out, its possible I just was printing before but I sort of doubt it.