micropython/micropython #10787: Fail to compile ESP32 port with HEAD 4598b89

mirokymac, 2023-02-19:
machine info:
 - arch: ARMv8
 - os: debian buster
 - esp-idf version: v4.4

pull the repo and compiling with:
```git clone https://github.com/micropython/micropython.git
cd micropython
git submodules updagte --init
make -C mpy-cross
cd ports/esp32
make submodules
make BOARD=GENERIC
```

then get error while compiling: `make[3]: *** No rule to make target 'genhdr/mpversion.h', needed by 'genhdr/qstr.i.last'.  Stop.
`

full log as the attachment.
[mpy_compilefailed_2023_2_19.log](https://github.com/micropython/micropython/files/10775985/mpy_compilefailed_2023_2_19.log)

I have try `git checkout v1.19.1` then pull submodules, the compile gets no error.

Any idea for fixing?

dpgeorge, 2023-02-21:
Can you try building at this commit: 35524a6fda1e44692ad599a39c802c168c897de9 (using IDF v4.4).

mirokymac, 2023-02-21:
@dpgeorge 355246a with IDF v4.4 can go through without any issue.

jimmo, 2023-02-22:
@mirokymac I dug into this, it appears that the fix we made in https://github.com/micropython/micropython/commit/9848b0685f3f24ec9fa7f179a1575b8a76565073 (see #10691) is incompatible with CMake when it's using `make` as the backend (rather than `ninja`).

(Yes, the ESP32 build goes: make -> cmake -> ninja/make)

Getting that fix to work was important and solves a big problem. It would be nice to make it work with `make` too, but unfortunately we don't have the capacity to investigate right now.

In the meantime, there is a simple fix: `apt install ninja-build`. Then if you do a `make clean` the build will subsequently just use `ninja` instead.

jimmo, 2023-02-22:
Update: as far as I can tell this problem goes away with later versions of CMake (buster has 3.13, I have 3.25 locally). On my PC I can only replicate this with the make generator when I downgrade to cmake 3.13, but works fine with cmake 3.25.

So my suspicion is that this was a bug in the Make generator, fixed in later CMake revisions.

closed 2023-02-22 by jimmo

mirokymac, 2023-02-22:
@jimmo thanks. I will lean some ninja afterward.

I also try to compile on AMD64 after I bug into the ARMv8 thing, and get no error. And as you said the CMake version is the key.

CMake on my ARM machine: 3.13.4

CMake on my PC: 3.18.4