micropython/micropython #10794: network.AP_IF creates a weird AP name and unable to connect

adriangalilea, 2023-02-20:
Raspberry pi pico w
Last u2f that downloads from [here](https://www.raspberrypi.com/documentation/microcontrollers/micropython.html#what-is-micropython)
relevant sidenote: have no idea why this downloads "unstable" versions or how to download stable ones.

When doing:
`import network

ap = network.WLAN(network.AP_IF)
ap.active(True)
ap.config(essid='pico', password='123456789')
`

it creates a network called PICOXXXXX(numbers and characters) instead of "pico". I tested in 2 different devices, unable to connect to that network either, seems like creating an AP on the pico w is broken.

dpgeorge, 2023-02-21:
You need to set the configuration before making the AP active.  Also you can use `ssid` instead of the old `essid` parameter.  Eg:
```py
ap = network.WLAN(network.AP_IF)
ap.config(ssid="pico", pasword="12345678")
ap.active(True)
```

adriangalilea, 2023-02-21:
That was it, sorry.

So the pico creates a default PICOXXXXX network, interesting, what would be the password in that instance?

closed 2023-02-21 by adriangalilea

dpgeorge, 2023-02-22:
> So the pico creates a default PICOXXXXX network, interesting, what would be the password in that instance?

The default password for that default AP is "picoW123".

Erkman84, 2023-04-16:
Hi guys, I can confirm, the issue still persists: v1.19.1-1014-gbde222ce8 this is what Thonny offers.
I cannot change any params also before active(True). I get the PICOXXXX and the password seems picoW123 maybe working, but also hangs and I cannot connect. This function is important for wireless projects, without this, it has less effort...