micropython/micropython #10799: pic16bit doesn't compile

Tico06, 2023-02-20:
Hi There,

I tried to compile the pic16bit port.

I installed xc16 v2.00, and updated the Makefile as following:

> 
> git diff Makefile
> diff --git a/ports/pic16bit/Makefile b/ports/pic16bit/Makefile
> index d2c8df470..1b546f6cf 100644
> --- a/ports/pic16bit/Makefile
> +++ b/ports/pic16bit/Makefile
> @@ -7,7 +7,7 @@ QSTR_DEFS = qstrdefsport.h
>  include $(TOP)/py/py.mk
>  include $(TOP)/extmod/extmod.mk
>  
> -XCVERSION ?= 1.35
> +XCVERSION ?= 2.00
>  XC16 ?= /opt/microchip/xc16/v$(XCVERSION)
>  CROSS_COMPILE ?= $(XC16)/bin/xc16-
>  
> @@ -18,6 +18,7 @@ INC += -I.
>  INC += -I$(TOP)
>  INC += -I$(BUILD)
>  INC += -I$(XC16)/include
> +INC += -I$(XC16)/include/bits
>  INC += -I$(XC16)/support/$(PARTFAMILY)/h
>  
>  CFLAGS_PIC16BIT = -mcpu=$(PART) -mlarge-code

When launching the make, I have the following output:

> 
> Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
> GEN build/genhdr/qstr.i.last
> GEN build/genhdr/qstr.split
> GEN build/genhdr/qstrdefs.collected.h
> QSTR updated
> GEN build/genhdr/qstrdefs.generated.h
> GEN build/genhdr/moduledefs.split
> GEN build/genhdr/moduledefs.collected
> Module registrations updated
> GEN build/genhdr/moduledefs.h
> GEN build/genhdr/root_pointers.split
> GEN build/genhdr/root_pointers.collected
> Root pointer registrations updated
> GEN build/genhdr/root_pointers.h
> mkdir -p build/extmod/
> mkdir -p build/py/
> mkdir -p build/shared/libc/
> mkdir -p build/shared/readline/
> mkdir -p build/shared/runtime/
> CC ../../py/mpstate.c
> In file included from ../../py/mpstate.h:34:0,
>                  from ../../py/mpstate.c:27:
> ../../py/nlr.h:109:5: error: expected specifier-qualifier-list before 'jmp_buf'
> In file included from ../../py/obj.h:33:0,
>                  from ../../py/mpstate.h:35,
>                  from ../../py/mpstate.c:27:
> ../../py/qstr.h:49:18: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'qstr_short_t'
> ../../py/qstr.h:52:17: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'qstr_hash_t'
> ../../py/qstr.h:60:17: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'qstr_len_t'
> ../../py/qstr.h:72:5: error: expected specifier-qualifier-list before 'qstr_hash_t'
> In file included from ../../py/mpstate.h:35:0,
>                  from ../../py/mpstate.c:27:
> ../../py/obj.h:601:5: error: expected specifier-qualifier-list before 'uint16_t'
> ../../py/obj.h:683:5: error: expected specifier-qualifier-list before 'uint16_t'
> ../../py/obj.h:704:5: error: expected specifier-qualifier-list before 'uint16_t'
> In file included from ../../py/mpstate.h:35:0,
>                  from ../../py/mpstate.c:27:
> ../../py/obj.h:1185:5: error: expected specifier-qualifier-list before 'uint32_t'
> In file included from ../../py/mpstate.c:27:0:
> ../../py/mpstate.h:112:5: error: expected specifier-qualifier-list before 'uint16_t'
> In file included from ../../py/mpstate.c:27:0:
> ../../py/mpstate.h:257:5: error: expected specifier-qualifier-list before 'uint16_t'
> See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
> make: *** [../../py/mkrules.mk:83 : build/py/mpstate.o] Erreur 1

Thanks a lot for your support.

Eric.

dpgeorge, 2023-02-21:
> +INC += -I$(XC16)/include/bits

I don't think you should add this include line.

Tico06, 2023-02-21:
Hi there,

Thanks a lot for your answer. 

Without out I am having the following error:

> ../../extmod/modbtree.c:30:19: fatal error: fcntl.h: No such file or directory

This file is available from different location on my HD:

> 
> /opt/microchip/xc16/v2.00/include/bits/fcntl.h
> /usr/arm-none-eabi/include/fcntl.h
> /usr/arm-none-eabi/include/sys/_default_fcntl.h
> /usr/arm-none-eabi/include/sys/fcntl.h
> /usr/avr/include/fcntl.h
> /usr/include/fcntl.h
> /usr/include/asm/fcntl.h
> /usr/include/asm-generic/fcntl.h
> /usr/include/bits/fcntl.h
> /usr/include/linux/fcntl.h
> /usr/include/sys/fcntl.h

The kernel provides also this file.

As I compile with XC16, I believed that was the one it's looking for.

regards,

Eric.

dpgeorge, 2023-02-21:
> Without out I am having the following error:
> 
> > ../../extmod/modbtree.c:30:19: fatal error: fcntl.h: No such file or directory

This should be fixed by 95082693c3cae9a7f2b9cac9a2473ead4146c3c8

Tico06, 2023-02-21:
Hi There,

Thanks a lot. It now compiles, but linking failed with the following

> LINK build/firmware.elf
> 
> Info: Loading file: .//opt/microchip/xc16/v2.00/support/dsPIC33F/gld/p33FJ256GP506.gld
> /opt/microchip/xc16/v2.00/bin/bin/elf-ld: cannot find -lc
> make: *** [Makefile:66 : build/firmware.elf] Erreur 255

I have no clue where to look..

Regards,

Eric.

Tico06, 2023-02-24:
Hi There,

I guess this is an environment configuration issue. I'll focus on other MCUs and get back to this issue in the forum later.

Thanks

Eric.

Tico06, 2023-02-24:
See last comment.

closed 2023-02-24 by Tico06