micropython/micropython #10812: Sockets not releasing for Pico_W

cascmptrski, 2023-02-22:
Using a very simple web server on the Pico_W, running it once works fine. Running it a second time throws an address in use error. The only way to reset the system is to remove power. Stopping the device (e.g. in Thonny) still gives the same error when attempting to run the code.

The expectation is that once the code stops and/or is reset that running the code again would provide the same response as the first time.

The MicroPython build: is rp2-pico-w-20230221-unstable-v1.19.1-887-gb11026689
Version Reported: MicroPython v1.19.1-887-gb11026689 on 2023-02-21; Raspberry Pi Pico W with RP2040

The first file is the simple web server, the python script 
[sweb-server-d.txt](https://github.com/micropython/micropython/files/10800292/sweb-server-d.txt)

Running the above script two successive times yields the following information in the terminal window
[repl-session.txt](https://github.com/micropython/micropython/files/10800295/repl-session.txt)

Additionally, the output does not display on the web site every time, but only occasionally

dpgeorge, 2023-02-22:
You need to close the waiting (bound on port 80) socket, otherwise you won't be able to make a new one on the same port.

A soft reset (ctrl-D at the REPL on a blank line) will automatically reclaim all sockets.

cascmptrski, 2023-02-22:
Actually, I have tried this multiple times and the only thing that resets the socket is a full power reset. The <cntl+D> will not effect a reset and release the socket, even after the code has finished running. Using the attached code will demonstrate the problem.

cascmptrski, 2023-02-22:
[repl-session1.txt](https://github.com/micropython/micropython/files/10806855/repl-session1.txt)

This file shows the problem even after executing the soft-reset via <cntl+d> at the command line.

cascmptrski, 2023-02-22:
This appears to be exactly the same as issue #3739 with the ESP8266. 

At this point it appears that the same suggestions will correct the problem:
Include the following line  prior to the socket.bind(addr) command

socket.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)

As the network stack is not being released, and according to the responses in the prior issue, it will not be fixed, the only way to release the stack is via a hard reset. This means either removing power or issuing the following command from a REPL prompt
machine.reset()
which effectively does the same thing

dpgeorge, 2023-02-22:
> At this point it appears that the same suggestions will correct the problem: Include the following line prior to the socket.bind(addr) command
> 
> socket.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)

Ah yes, you are correct.  Since a previous socket was bound to that port, the port will remain in use for a certain time even if the socket is closed.  The reason is, to allow any outstanding packets/connections to still come in to the MCU and be processed correctly (eg so the remote peer can close its TCP socket cleanly).

Using `SO_REUSEADDR` is the way to make it work.  That's also what's needed on CPython (on Linux at least).

> As the network stack is not being released, and according to the responses in the prior issue, it will not be fixed, the only way to release the stack is via a hard reset

You shouldn't need to do a hard reset at all.  If you close all sockets and use `SO_REUSEADDR` then the same code should be able to run again and again.

jonnor, 2024-09-02:
A proposed fix has been suggested above, but has not been confirmed by tester. Please see if the suggestion fixes the problem.

closed 2025-03-03 by jonnor

ggeorg31, 2025-09-06:
i had the same problem and machine.reset() worked for me