micropython/micropython #10813: Look for modules in every directory of `usys.path`

smurfix, 2023-02-22:
Current behavior: if `usys.path==["/lib",".flash"]` and there's a `/lib/foo/` directory, `import foo.bar` will fail even though `help("modules") does list it and `.flash` still is in `usys.path`.

This is a problem because when I write and debug large programs which don't fit into RAM, I often want to deploy new versions of single modules without disrupting everything else. Right now I have to re-flash the whole board just to add one `print()` line somewhere. This also raises concerns about the number of write cycles.

Proposed fix: teach MicroPython to check all elements of `usys.path` for the relevant file-or-directory, instead of stopping at the first one it finds.

dpgeorge, 2023-02-22:
See related #8922

smurfix, 2023-02-22:
Thanks for the pointer.

Comparing the two approaches: this PR doesn't have "allow_frozen" flags or the need to specify a `frozen_overlay_path`; I don't see the need for either.

Specifically: You don't want an overlay? Simply don't add it to `usys.path`. Also I don't see a reason for skipping a FS module just because it's not in the frozen set; on the other hand, when I need to add some code to the FS, I don't see why it should be required to do this as a new top-level module OR to add a dummy debug module to the frozen set.