micropython/micropython #10867: uasyncio StreamReader fails if UART times out.

peterhinch, 2023-02-26:
I'm unsure if this qualifies as a bug as there is an obvious workround.

The issue arose in [this report](https://github.com/peterhinch/micropython-async/issues/99). If the UART receiver times out the device returns `None` which causes `uasyncio` to fail [here](https://github.com/micropython/micropython/blob/2e4dda3c2072c4de15ac8d51ddb471b9c3833db7/extmod/uasyncio/stream.py#L66).

Carglglz, 2023-02-26:
I bump into that error too and this  https://github.com/micropython/micropython/pull/8968/commits/6f9a282ea5687457ef87b3fb66694e7f9ce97e7a fixes it

Zuikkis, 2023-02-27:
I'm the one that made the original issue report in uasyncio git.

Interestingly, at least for Raspberry Pi Pico, timeout defaults to 0ms and timeout_char also defaults to 0ms but is then incremented slightly to make it longer than one character depending on the baudrate.

https://github.com/micropython/micropython/blob/master/ports/rp2/machine_uart.c

Doesn't this mean that machine_uart_read() will always fail unless there's already bytes in the ring buffer? 

Anyway, my problem was fixed by explicitly adding "timeout=100000, timeout_char=100000" when opening the UART.

shodan8192, 2023-03-02:
I'm also faced this error in my program, but only when pico is overclocked above ~190MHz and baudrates in range about 80000-150000bps

To reproduce, this code and connection between RX and TX are sufficient, it takes few seconds to crash :

```python
import uasyncio as asyncio
from machine import UART, freq

TEST_STR = b"TEST-0123456789\n"

freq(200_000_000)

uart = UART(1, 115200)
uart_reader = asyncio.StreamReader(uart)
uart_writer = asyncio.StreamWriter(uart)
received = asyncio.Event()

async def writer():
    while True:
        uart_writer.write(TEST_STR)
        await uart_writer.drain()
        await received.wait()
        received.clear()

async def main():
    asyncio.create_task(writer())

    while True:
        line = await uart_reader.readline()
        print(line)
        received.set()

asyncio.run(main())
```

In this case (and possibly others), the cause is here : https://github.com/micropython/micropython/blob/master/ports/rp2/machine_uart.c#L549
I think that check of `uart_is_readable` shouldn't be here, because it may return True while ringbuf is not yet filled by interrupt service, which in turn triggers readline from an empty buffer. If there's only `ringbuf_avail` check, program works as expected.

peterhinch, 2023-04-10:
This has cropped up again [here](https://github.com/peterhinch/micropython-async/issues/102) on code which runs without issue on Pyboards. I think (but have not proved) that the GPS data source occasionally has inter-character gaps substantially longer than one character period.

There is a generic question for UARTs: what should the default timeout be? There is a case to be made for infinity. The default on Pico seems extremely short. If the default is to be finite, then `uasyncio` should be capable of dealing with the case.

peterhinch, 2023-06-01:
This keeps cropping up in [Discussions](https://github.com/orgs/micropython/discussions/11649#discussioncomment-6057841).

We need a way to disable UART timeouts because `uasyncio` can apply any timeouts required by the application. There is no "right" value for a UART timeout as some devices only speak when spoken to. You might have a readline lasting days...