micropython/micropython #10870: ESP32S3 Accesspoint (AP) issue

StevenSlaa, 2023-02-26:
Hello everyone,

I have an ESP32S3 board ([FeatherS3](https://esp32s3.com/feathers3.html)). I am trying to create an simple accesspoint, however, for some reason I can't get it to work (can't find the FeatherS3 in list on my phone or PC). I used the following code:
```python
import network
ap = network.WLAN(network.AP_IF)

#  (the original parameter from the documentation is 'ssid' this seems the be wrong. Also with the password parameter it does not work)
ap.config(essid='FeatherS3')

ap.config(max_clients=10)
ap.active(True)
```
**References**: 
- [Official Documentation](https://docs.micropython.org/en/latest/esp32/quickref.html#wlan)
- [RandomNerdTutorials](https://randomnerdtutorials.com/micropython-esp32-esp8266-access-point-ap/)
- [Issue 4185](https://github.com/micropython/micropython/issues/4158)
- [ESP32 Accesspoint in Micropython (TechToTinker)](https://www.youtube.com/watch?v=LKQvzN_Rk-U&ab_channel=TechToTinker)

**Firmware**: [v1.19.1 (2022-06-18)](https://micropython.org/download/feathers3/) (Flashed according to the instructions)
**Power Supply**: Default USB-C connector with a short cable to a stable power brick.

My original plan is to create an accesspoint and connect to the **webrepl**, but first things first. As mentioned earlier not even a basic accesspoint is working right now, and never was. so let's hopefully fix that first.

## Result

[image]

jonnor, 2024-09-01:
Does this still happen on latest MicroPython? Version 1.23 or later. There have been several improvement regarding ESP32S3.

StevenSlaa, 2024-09-02:
Thank ypu fpr your comment. Completely forgot about this issue. I will give it a shot and close the issue when it works.👍

StevenSlaa, 2024-09-03:
Confirmed, that accesspoint is now working correctly. WebREPL sadly does not. Anyway, thats a different issue. This can now be closed.

closed 2024-09-03 by StevenSlaa