micropython/micropython #10872: AttributeError: 'module' object has no attribute 'sockaddr'

klint-k, 2023-02-27:
As not seen in the code the connection works. When the Error happens is when the webpage is to be loaded.

Version:
MicroPython v1.19.1-896-g2e4dda3c2 on 2023-02-25; Raspberry Pi Pico W with RP2040

Error:
File "/lib/socket.py", line 47, in accept
AttributeError: 'module' object has no attribute 'sockaddr'

Code:
```
import socket
from do_connect import * #  to do the connection
from test_ssid import * # to check if the ssid is there

if test_ssid:
    do_connect()
    
with open("index.html", "r") as page:
    html = page.read()

addr = socket.getaddrinfo('0.0.0.0', 80)[0][-1]
s = socket.socket()
s.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)
s.bind(addr)
s.listen(1)

while True:
    cl, addr = s.accept()
   ...

```

Code from socket:
```
43  class socket(_socket.socket):
44  
45      def accept(self):
46          s, addr = super().accept()
**47          addr = _socket.sockaddr(addr)**
48          return (s, (_socket.inet_ntop(addr[0], addr[1]), addr[2]))

```

klint-k, 2023-02-27:
Well here is the fix.
```

    def accept(self):
        s, addr = super().accept()
        addr = _socket.getaddrinfo(addr[0], addr[1])[0] ###  add
        #addr = _socket.sockaddr(addr) ### remove
        #return (s, (_socket.inet_ntop(addr[0], addr[1]), addr[2])) ### remove
        return (s, addr[-1])
```

dpgeorge, 2023-02-27:
> File "/lib/socket.py", line 47, in accept

Did you install `socket.py` from micropython-lib?

Try removing `lib/socket.py` and then things should work.

closed 2026-06-04 by dpgeorge