micropython/micropython #10876: The yield test result is abnormal

lbuque, 2023-02-27:
I'm developing something with esp32-s3 and I need to use yield.

I ran the code and it didn't match the documentation's test results.

[I ran the code and it didn't match the documentation's test results.](https://docs.micropython.org/en/v1.19.1/genrst/core_language.html?highlight=yield#generator)

[image]

The precompiled firmware I used is: https://micropython.org/resources/firmware/GENERIC_S3-20220618-v1.19.1.bin

Is this a bug, or am I using it incorrectly?

BrianPugh, 2023-02-27:
it'd be better to upload the code rather than a screenshot so others can test. Here's a text version of the code:

```
class Foo:
    def __enter__(self):
        print("enter")

    def __exit__(self, *args):
        print("exit")

def bar(x):
    with Foo():
        while True:
            x += 1
            yield x

def func():
    g = bar(0)
    for _ in range(3):
        print(next(g))

func()
```

python output:
```
$ python ibuque.py
enter
1
2
3
exit
```

micropython (latest unix) output:
```
$ micropython ibuque.py
enter
1
2
3
```

The `exit` might not be there because it might need to be garbage collected or something, but seems like the code works?

lbuque, 2023-02-28:
@BrianPugh But I can't get normal results using micropython of esp32-s3!

peterhinch, 2023-02-28:
There are two issues here:
 1. Whether the code runs on ESP32-S3
 2. Whether it is reasonable to expect a context manager to work in this scenario.

Re 1. I cannot replicate this fault. Testing a UM Feather S3 with `mpremote`:
```python
MicroPython v1.19.1-782-g699477d12 on 2022-12-20; FeatherS3 with ESP32-S3
Type "help()" for more information.
```
Result:
```
enter
1
2
3
```
Might this be a Thonny issue?

Re 2, consider this pattern: 
```python
def foo():
    with some_context_manager:
        while True:
            yield something
g = foo()
```
The generator `g` returns an infinite set. In an application `g` might be called from multiple places. The generator never terminates so it is not clear to me how the CM's `__exit__` method can ever be called. I guess CPython does this on exit from the interpreter, but most embedded applications run forever.

lbuque, 2023-02-28:
@peterhinch I updated thonny to 4.0.2 and everything works fine.
Thanks a lot for your answer!
I will close this issue.

closed 2023-02-28 by lbuque