micropython/micropython #10878: tools/mpremote: "Error getting the version from source" when installing mpremote from source

mdaeron, 2023-02-27:
I cloned the latest `master` branch and tried to install `mpremote` from source:

```
git clone https://github.com/micropython/micropython.git
cd micropython/tools/mpremote
python -m pip install .
```

I get the following error:

```
Processing /Users/daeron/ShareBox/git/micropython/tools/mpremote
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Preparing metadata (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [34 lines of output]
      /private/var/folders/4l/bf7dw9v56dl9vxk0bvh3szvm0000gn/T/pip-build-env-7ejftfsw/overlay/lib/python3.11/site-packages/setuptools_scm/version.py:84: UserWarning: tag '0.0' no version found
        warnings.warn(f"tag {tag!r} no version found")
      Traceback (most recent call last):
        File "/Users/daeron/mambaforge/envs/mpr/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
          main()
        File "/Users/daeron/mambaforge/envs/mpr/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/Users/daeron/mambaforge/envs/mpr/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 152, in prepare_metadata_for_build_wheel
          whl_basename = backend.build_wheel(metadata_directory, config_settings)
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/private/var/folders/4l/bf7dw9v56dl9vxk0bvh3szvm0000gn/T/pip-build-env-7ejftfsw/overlay/lib/python3.11/site-packages/hatchling/build.py", line 56, in build_wheel
          return os.path.basename(next(builder.build(wheel_directory, ['standard'])))
                                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/private/var/folders/4l/bf7dw9v56dl9vxk0bvh3szvm0000gn/T/pip-build-env-7ejftfsw/overlay/lib/python3.11/site-packages/hatchling/builders/plugin/interface.py", line 93, in build
          self.metadata.validate_fields()
        File "/private/var/folders/4l/bf7dw9v56dl9vxk0bvh3szvm0000gn/T/pip-build-env-7ejftfsw/overlay/lib/python3.11/site-packages/hatchling/metadata/core.py", line 243, in validate_fields
          _ = self.version
              ^^^^^^^^^^^^
        File "/private/var/folders/4l/bf7dw9v56dl9vxk0bvh3szvm0000gn/T/pip-build-env-7ejftfsw/overlay/lib/python3.11/site-packages/hatchling/metadata/core.py", line 128, in version
          self._version = self._get_version()
                          ^^^^^^^^^^^^^^^^^^^
        File "/private/var/folders/4l/bf7dw9v56dl9vxk0bvh3szvm0000gn/T/pip-build-env-7ejftfsw/overlay/lib/python3.11/site-packages/hatchling/metadata/core.py", line 222, in _get_version
          core_metadata = self.core
                          ^^^^^^^^^
        File "/private/var/folders/4l/bf7dw9v56dl9vxk0bvh3szvm0000gn/T/pip-build-env-7ejftfsw/overlay/lib/python3.11/site-packages/hatchling/metadata/core.py", line 170, in core
          self._version = self._get_version(metadata)
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/private/var/folders/4l/bf7dw9v56dl9vxk0bvh3szvm0000gn/T/pip-build-env-7ejftfsw/overlay/lib/python3.11/site-packages/hatchling/metadata/core.py", line 226, in _get_version
          version = self.hatch.version.cached
                    ^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/private/var/folders/4l/bf7dw9v56dl9vxk0bvh3szvm0000gn/T/pip-build-env-7ejftfsw/overlay/lib/python3.11/site-packages/hatchling/metadata/core.py", line 1412, in cached
          raise type(e)(message) from None
      AssertionError: Error getting the version from source `vcs`:
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a proble
m with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
```

Also, when I try `python -m mpremote --version` after this error, I get `mpremote 0.3.0`. Yet the current PyPI version is '0.4.0'

Is there a better way to install the latest version of `mpremote`?

dlech, 2023-02-27:
python -m pip install https://github.com/micropython/micropython/archive/refs/heads/master.zip

?

mdaeron, 2023-02-27:
Thanks, that works. For others interested in doing so in the future, the latest build artifact can be found [here](https://github.com/micropython/micropython/actions/workflows/mpremote.yml).

closed 2023-02-27 by mdaeron