micropython/micropython #10895: core; Crash with  an example code with walrus use,  on V1.19 ?

robert-hh, 2023-03-01:
### Discussed in https://github.com/orgs/micropython/discussions/10879

<div type='discussions-op-text'>

<sup>Originally posted by **fca1** February 28, 2023</sup>
Hi All,!

I've encourated a strange behavior with micropython 1.19 (crash and reset), my target is an ESP32 with spiram. 
The piece of code to attempt to reproduce that, is very tiny. This code has been tested with cpython, without any problems. 

Please, if someone could confirm if there is really a problem ? Thanks a lot in advance, 

def for_test_purpose():
    lst= [ 0,1]
    v=0
    x=1  # crash if  this line is present. 
    print([v := i for i in lst if i > v])

if __name__ == "__main__":
    for_test_purpose()    
    </div>

dpgeorge, 2023-03-02:
Thanks for the report.  I can confirm the crash on the unix port with latest master.

It's a nasty bug, it's not computing the closure of `v` correctly in the list comprehension.

dpgeorge, 2023-03-02:
See #10902 for a fix.

dpgeorge, 2023-03-09:
Fixed by 7c1584aef10f80aabfeba242892ec5f85c57bead

closed 2023-03-09 by dpgeorge