micropython/micropython #10898: Interrupting soft reset (Ctrl-D) with Ctrl-C erases filesystem on Raspberry Pi Pico

connorcl, 2023-03-01:
Sending a Ctrl-D to the MicroPython REPL to initiate a soft reset and then immediately issuing a Ctrl-C causes the entire filesystem to be erased on the Raspberry Pi Pico board. This issue was observed on 2 separate Pico boards across the following nightly builds (the 4 latest as of 01 March 2023):

- [v1.19.1-910-g4937174b4](https://micropython.org/resources/firmware/rp2-pico-20230228-unstable-v1.19.1-910-g4937174b4.uf2)
- [v1.19.1-909-g36ae5d3e8](https://micropython.org/resources/firmware/rp2-pico-20230228-unstable-v1.19.1-909-g36ae5d3e8.uf2)
- [v1.19.1-900-gfc4c47f7b](https://micropython.org/resources/firmware/rp2-pico-20230227-unstable-v1.19.1-900-gfc4c47f7b.uf2)
- [v1.19.1-897-g72e931832](https://micropython.org/resources/firmware/rp2-pico-20230226-unstable-v1.19.1-897-g72e931832.uf2)

However, the bug was not observed in the latest stable version 1.19.1.

### Steps to reproduce

These steps were tested on a Linux host (Ubuntu 22.04).

First, load one of the above nightly versions of MicroPython onto a Raspberry Pi Pico board.

Next, using `mpremote`, install a library on the Pico’s filesystem, for example hashlib:

```
mpremote mip install hashlib
```

Then check that the library exists on the Pico’s filesystem:

```
mpremote fs ls /lib
```

This should show the `hashlib` folder:

```
ls :/lib
           0 hashlib/
```

Then, to send the required keystrokes (Ctrl-D followed by Ctrl-C), execute the following lines of Python on the Linux host machine to which the Pico is connected, replacing `/dev/ttyACM0` with the serial port of the MicroPython REPL. This script requires [pyserial](https://pypi.org/project/pyserial/).

```
import serial
s = serial.Serial('/dev/ttyACM0')
s.write(b'\x04') # Ctrl-D
s.write(b'\x03') # Ctrl-C
s.close()
```

Less precisely, the issue can also be reproduced by entering a REPL with `mpremote` and pressing Ctrl-D and Ctrl-C in rapid succession.

Then check the Pico’s filesystem once again:

```
mpremote fs ls
```

This should show that the `lib` folder and all files it contains have been removed:

```
ls :
```

This issue became evident because the visual studio code extension [Pico-W-Go](https://github.com/paulober/Pico-W-Go) exhibits this behaviour when running a MicroPython file, issuing a Ctrl-D and Ctrl-C in rapid succession before running the given code. This issue has been reported on that project’s GitHub page [here](https://github.com/paulober/Pico-W-Go/issues/46), suggesting it affects other nightly versions older than the 4 listed above.

robert-hh, 2023-03-01:
See PR #10833

connorcl, 2023-03-01:
Aha thanks, couldn't find an issue for this but didn't check open PRs!