micropython/micropython #10900: wifi.scan() requires sta.active(False) before sta.active(True)

davefes, 2023-03-01:
### Discussed in https://github.com/orgs/micropython/discussions/10886

<div type='discussions-op-text'>

<sup>Originally posted by **wh33t** March  1, 2023</sup>
So my apologies upfront if this whole discussion is bogus due to my noobness. 

I am completely unfamiliar with ESP32 and Python ... I'm just trying to get my feet wet with the fundamentals.

I am using the Thing Plus C ESP32 from Sparkfun and the Thonny Editor.

I just want to power up the ESP32, and scan to see which wifi networks it can see.

This code will not produce any output when run.

```
# wifi_test.py
import network
import time

# Create a station object to store our connection
wifi = network.WLAN(network.STA_IF)
wifi.active(True)  

available_wifi_networks = wifi.scan()

for i in available_wifi_networks:
    print (i[0] + " " + i[1])
```

However, I have noticed that if I declare the wifi object, set it's active status to false, sleep the device for 1 second, and then set it back to active it will work as expected and reliably so.

```
# wifi_test.py
import network
import time

# Create a station object to store our connection
wifi = network.WLAN(network.STA_IF)
wifi.active(False)  
time.sleep(1)
wifi.active(True)  

available_wifi_networks = wifi.scan()

for i in available_wifi_networks:
    print (i[0] + " " + i[1])
```
Is this ESP32 unit from Sparkfun faulty? Am I doing something wrong here?

Any and all help greatly appreciated. Cheers.</div>

davefes, 2023-03-01:
Doing a wifi.scan() seems to require that the sta interface be in a known state before proceeding.  This implementation of wifi.py
https://github.com/glenn20/micropython-espnow-utils/tree/main/src
and in def reset() appears to insure that this condition is met.

Could the existing `def do_connect()` example at : https://docs.micropython.org/en/latest/esp32/quickref.html
be modified or the under-lying calls to ESP-IDF be changed so that connection is handled is a better way?

zcattacz, 2023-05-31:
In my recent case, power is the cause, wifi is setup in boot.py, both wifi.active(True) and wifi.scan() could sometimes randomly cause a reset on boot, after replacing the 3V3/GND jumper wire and replaced the external LDO with a DCDC converter. the reset is gone.

closed 2026-05-25 by dpgeorge